espressif/esp-at #688: HTTPCGET max size

fabien-lunii, 2022-12-09:
Hello,
I do HTTPCGET to download a file (17 Mbytes) but it always stop aroung 512 Kbytes. Is there any limitation ?

Thank you.

fabien-lunii, 2022-12-09:
To test: AT+HTTPCGET="https://leolab-dev.fr/projets/test/file_example_MP4_1920_18MG.mp4",2048,2048,15000

ustccw, 2023-02-07:
@fabien-lunii it doesn't seem to work even on my pc if download url is `https` and download speed is in a low level.
[image]. maybe the timeout limit or other factors on server side?

i tested it on esp32-c3 chip, seem that `https` connection always stops around several MB. 
And tried to use `http` instead of `https`, seem to work.

closed 2023-06-05 by ustccw