espressif/esp-at #689: esp32c3 mini-1 AT

zjh1097900564, 2022-12-15:
I changed the default IO of uart0 to 21 and 20, because this module is designed to be 21 and 20, but there is no response to any AT instruction. Power on and out the following log again

RX：ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x1a04
load:0x403ce000,len:0x958
load:0x403d0000,len:0x2de0
entry 0x403ce000
[0;32mI (32) boot: ESP-IDF v4.3.1-354-g98d34e5f6d 2nd stage bootloader[0m
[0;32mI (32) boot: compile time 19:19:39[0m
[0;32mI (32) boot: chip revision: 3[0m
[0;32mI (35) boot.esp32c3: SPI Speed      : 40MHz[0m
[0;32mI (40) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (44) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Enabling RNG early entropy source...[0m
[0;32mI (55) boot: Partition Table:[0m
[0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (65) boot:  0 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  2 nvs              WiFi data        01 02 00010000 0000e000[0m
[0;32mI (88) boot:  3 at_customize     unknown          40 00 0001e000 00042000[0m
[0;32mI (95) boot:  4 ota_0            OTA app          00 10 00060000 001d0000[0m
[0;32mI (103) boot:  5 ota_1            OTA app          00 11 00230000 001d0000[0m
[0;32mI (110) boot: End of partition table[0m
[0;32mI (115) esp_image: segment 0: paddr=00060020 vaddr=3c140020 size=29040h (168000) map[0m
[0;32mI (158) esp_image: segment 1: paddr=00089068 vaddr=3fc90800 size=03dfch ( 15868) load[0m
[0;32mI (162) esp_image: segment 2: paddr=0008ce6c vaddr=40380000 size=031ach ( 12716) load[0m
[0;32mI (167) esp_image: segment 3: paddr=00090020 vaddr=42000020 size=130d20h (1248544) map[0m
[0;32mI (434) esp_image: segment 4: paddr=001c0d48 vaddr=403831ac size=0d588h ( 54664) load[0m
[0;32mI (447) esp_image: segment 5: paddr=001ce2d8 vaddr=50000000 size=00004h (     4) load[0m
[0;32mI (448) esp_image: segment 6: paddr=001ce2e4 vaddr=50000008 size=00010h (    16) load[0m
[0;32mI (458) boot: Loaded app from partition at offset 0x60000[0m
[0;32mI (459) boot: Disabling RNG early entropy source...[0m
module_name:MINI-1

max tx power=78,ret=0

2.3.0

ustccw, 2023-02-07:
@zjh1097900564 tbh, i am not sure what you need. ESP32-C3 frimware uses GPIO20 as UART0 RX, GPIO21 as UART0 TX by default: https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Get_Started/Hardware_connection.html#esp32-c3-series, and all logs will output from UART0, as what your firmware does.

If you wanna use GPIO20 and GPIO21 as UART1 (send/response AT commands), you could use the [at.py](https://github.com/espressif/esp-at/blob/master/tools/at.py) script to generate a new firmware what you want, the command like the following:
```
python tools/at.py -in ESP32-C3-MINI-1_AT_Bin_V2.4.2.0/ESP32-C3-MINI-1-V2.4.2.0/factory/factory_MINI-1.bin -tx 21 -rx 20
```

zjh1097900564, 2023-02-07:
Thank you very much. Problem solved. It's my problem. I missed one change

zjh1097900564, 2023-02-07:
Thank you very much. Problem solved. It's my problem. I missed one change

赵镜涵--(赵磊)
***@***.***

ustccw, 2023-02-07:
no worries

closed 2023-04-20 by ustccw