espressif/esp-at #692: ESP32-AT v2.4.0.0 menuconfig error

zhoukeroro, 2023-01-09:
Hi everyone. I have a problem when compiling the ESP-AT v2.4.0.0 project.

I post the detail on esp32-at forum:
[https://www.esp32.com/viewtopic.php?f=34&t=31453](url)

I found there are so many developers meet the related issue (CMake Error), but there is no clear answer to solve it.

Could anybody help me to solve it? 

Best regards.

zhoukeroro, 2023-01-09:
the log says _'CMake Error at components/customized_partitions/CMakeLists.txt:40 (list):
list index: 1 out of range (-1, 0)'_, but I could not find where the [list index] come from.

> silence:0 , ld core:D:/esp/gitee/esp-at2/components/at/lib/libesp32_at_core.a
Parsing CSV input...
Verifying table...
CMake Error at components/customized_partitions/CMakeLists.txt:40 (list):
list index: 1 out of range (-1, 0)

zhoukeroro, 2023-01-10:
@xcguang, Sorry to disturb you. Could you HELP me?

xcguang, 2023-01-10:
@ustccw Please help with this.

ustccw, 2023-01-11:
@zhoukeroro could you please provide all detailed logs after you cloned esp-at project.

zhoukeroro, 2023-01-11:
> @zhoukeroro could you please provide all detailed logs after you cloned esp-at project.

hi, @ustccw , I post the all detail logs here:
[esp-at compile log.txt](https://github.com/espressif/esp-at/files/10389120/esp-at.compile.log.txt)

PS1: 'ESP-IDF 4.3 CMD' run in administrators mode.

ustccw, 2023-01-12:
@zhoukeroro 
1. 编译 hello world 工程时候，能用 idf.py menuconfig 配置吗。如果不能，可能要先解决这个。
2. 如果编译 hello world 工程可以使用 idf.py menuconfig 配置，但编译 esp-at 工程不能使用 python build.py menuconfig。那可以试试下面命令：
  ```
  cd esp-idf
  install.bat
  export.bat
  cd ..
  python build.py menuconfig
  ```

zhoukeroro, 2023-01-13:
@ustccw ，
1.按我上述相同的步骤，在你的环境下可以正常编译吗？
2.基于出错log的提示（CMake Error），能看出是大方向上是哪方面的问题吗？

ustccw, 2023-01-13:
1. 我这边是可以正常编译的。windows10 64bit。
2. 看起来是解析分区表出错了。
出错在这个附近：
https://github.com/espressif/esp-at/blob/release/v2.4.0.0/components/customized_partitions/CMakeLists.txt#L37
可以在 cmake 里加一点打印，看看哪个参数错误。比如：
```
    string(STRIP ${partition_args} args_value)
    string(REPLACE " " ";" args_lists ${args_value})
    list(GET args_lists 0 offset)
    list(GET args_lists 1 file_path)
    STRING(REGEX REPLACE ".*[/\\](.*)\.bin" "\\1" file_name ${file_path})
```
修改为：
```
    message("x1:${partition_args}\r\n x2:${customized_partition_args}\r\n")
    string(STRIP ${partition_args} args_value)
    message("x3:${args_value}\r\n")
    string(REPLACE " " ";" args_lists ${args_value})
    message("x4:${args_lists}\r\n")
    list(GET args_lists 0 offset)
    message("x5:${offset}\r\n")
    list(GET args_lists 1 file_path)
    message("x6:${file_path}\r\n")
    STRING(REGEX REPLACE ".*[/\\](.*)\.bin" "\\1" file_name ${file_path})
    message("x7:${file_name}\r\n")
```

正常打印应该是这样的：
```
Parsing CSV input...
Verifying table...
x1:0x21000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin 12K
 x2:0x21000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin 12K;0x24000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin 8K;0x26000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin 8K;0x28000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_ca.bin 8K;0x2a000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_cert.bin 8K;0x2c000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_key.bin 8K;0x2e000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_ca.bin 8K;0x37000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_cert.bin 8K;0x39000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_key.bin 8K;0x3B000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_ca.bin 8K;0x30000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/factory_param.bin 4K

x3:0x21000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin 12K

x4:0x21000;/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin;12K

x5:0x21000

x6:/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin

x7:ble_data

x1:0x24000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin 8K
 x2:0x21000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin 12K;0x24000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin 8K;0x26000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin 8K;0x28000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_ca.bin 8K;0x2a000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_cert.bin 8K;0x2c000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_key.bin 8K;0x2e000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_ca.bin 8K;0x37000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_cert.bin 8K;0x39000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_key.bin 8K;0x3B000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_ca.bin 8K;0x30000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/factory_param.bin 4K

x3:0x24000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin 8K

x4:0x24000;/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin;8K

x5:0x24000

x6:/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin

x7:server_cert

x1:0x26000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin 8K
 x2:0x21000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin 12K;0x24000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin 8K;0x26000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin 8K;0x28000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_ca.bin 8K;0x2a000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_cert.bin 8K;0x2c000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_key.bin 8K;0x2e000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_ca.bin 8K;0x37000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_cert.bin 8K;0x39000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_key.bin 8K;0x3B000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_ca.bin 8K;0x30000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/factory_param.bin 4K

x3:0x26000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin 8K

x4:0x26000;/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin;8K

x5:0x26000

x6:/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin

x7:server_key

[... 79 lines cut ...]
 x2:0x21000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/ble_data.bin 12K;0x24000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_cert.bin 8K;0x26000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_key.bin 8K;0x28000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/server_ca.bin 8K;0x2a000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_cert.bin 8K;0x2c000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/client_key.bin 8K;0x2e000 /home/chenwu/esp/at/ [... cut ...] ized_partitions/mqtt_key.bin 8K;0x3B000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/mqtt_ca.bin 8K;0x30000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/factory_param.bin 4K

x3:0x30000 /home/chenwu/esp/at/esp32c3-at/build/customized_partitions/factory_param.bin 4K

x4:0x30000;/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/factory_param.bin;4K

x5:0x30000

x6:/home/chenwu/esp/at/esp32c3-at/build/customized_partitions/factory_param.bin

x7:factory_param
```

zhoukeroro, 2023-01-13:
@ustccw  ，在cmake 里加上打印后的log见附件，看起来是路径解析出错了？
[esp-at compile log(add cmake log).txt](https://github.com/espressif/esp-at/files/10411919/esp-at.compile.log.add.cmake.log.txt)

zhoukeroro, 2023-01-14:
@ustccw ,问题解决了，路径解析出错跟我的PC安装了加密软件有关。关闭加密软件后就正常了。
万分感谢。

closed 2023-01-14 by zhoukeroro