espressif/esp-at #695: Blufi not returning SSID list

matt-wood-ct, 2023-02-07:
Hi we are trailing use of the blufi system for a product and we have found that with the current release ESP-AT firmware on a ESP32-C3 module we are not getting an SSID list back when running that blufi command, it is not returning any errors just also no SSIDs

Is this a known issue with the AT fimware or is it an issue with the app?

Many Thanks,
Matt

matt-wood-ct, 2023-02-08:
Further detail, we are sending a "Get the Wi-Fi list" command (0x9) and we are not receiving the followup frames as suggested in the documentation.
[image]
This issue is happening with both our own javascript implementation and the blufi demo app provided by Espressif.

Luxin812, 2023-02-10:
> Hi we are trialing use of the blufi system for a product and we have found that with the current release ESP-AT firmware on a ESP32-C3 module we are not getting an SSID list back when running that blufi command, it is not returning any errors just also no SSIDs
> 
> Is this a known issue with the AT fimware or is it an issue with the app?
> 

> Many Thanks, Matt

Hi,
Please download the AT firmware on the latest master branch from github to test this function. This function is only supported on AT not long ago. There is no release version that supports this function.
[image]

closed 2023-06-05 by ustccw