espressif/esp-at #699: connect to AWS MQTT broker

aviasaf, 2023-03-16:
Hi,
I am facing with a problem that happens from time to time.
I am trying to connect AWS MQTT broker,  but I receive '+MQTTDISCONNECTED:0'

I don't know the reason, but attached is a log :
module_name:WROOM-32
max tx power=78,ret=0
2.4.0
BLUFI BLE is not connected yet
all parameters are ready
ca len:1187
cert len:1220
key len:1679
E (16073) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
E (16073) esp-tls: Failed to open new connection
E (16074) TRANS_SSL: Failed to open a new connection
E (16084) MQTT_CLIENT: Error transport connect
MQTT ERROR
MQTT DISCONNECTED

As you can see, there is an error, what does it mean and I can I solve it?
 
The sequence that I am running from tera-term is:
:start
sendln 'AT+RST'
waitln 'OK' 'WIFI DISCONNECT'
pause (3)
sendln 'ATE0'
waitln 'OK'
pause (0)
sendln 'AT+SYSLOG=1'
waitln 'OK'
pause  (1)
sendln 'AT+CWMODE=1'
waitln 'OK'
pause (1)
sendln 'AT+CWJAP="asas_f1","xxxxxx"'
waitln 'OK' 'ready'
pause (0)
:mqttError
pause (5)	
sendln 'AT+MQTTUSERCFG=0,5,"esp32","espressif","1234567890",0,0,""'
waitln 'OK'
pause (3)
sendln 'AT+MQTTCONN=0,"xxxxxxxxx-ats.iot.eu-central-1.amazonaws.com",8883,1'
wait '+MQTTCONNECTED' '+MQTTDISCONNECTED' 'ERROR'
if result=1 goto mqttOK
if result=2 goto mqttError
if result=3 goto mqttError
:mqttOK 
pause (2)
sendln 'AT+MQTTCLEAN=0'
waitln 'OK'
goto start
sendln 'AT+MQTTSUB=0,"esp32/espressif",1'
waitln 'OK'
while (1)
sendln 'AT+MQTTPUB=0,"esp32/espressif","hello12345",0,0'
waitln 'OK'
pause (3)
endwhile
Thanks, in advance
Avi

aviasaf, 2023-04-01:
> @aviasaf <https://github.com/aviasaf> Have you followed the example here
> to complete all the steps.
>
> https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/mqtt-at-examples-for-cloud.html

Yes, my code is based on this example
Avi
בתאריך יום ו׳,

ustccw, 2023-04-03:
Could you please enable the debug log and see what happened.
- `./build.py menuconfig` -> `Component config` -> `Log output` -> `Default log verbosity` -> `Debug`
- `./build.py menuconfig` -> `Component config` -> `mbedTLS` -> `Enable mbedTLS debugging` -> `Set mbedTLS debugging level` -> `Verbose`

ustccw, 2023-04-03:
and could you please use other mqtt client tools to test if the configuration is ready for connection.
such as `mosquitto`.

closed 2023-06-05 by ustccw