espressif/esp-at #702: mbedtls ChaCha20 encryption on UART gets corrupted

LonesomeUnixCowboy, 2023-04-11:
I'm want to encrypt the UART communcation between esp-at and my MCU and it fails on the esp-at side. The data thats going out from esp-at (via at_port_write_data() callback in at_uart_task.c) gets a few bytes corrupted when there is a high through put of data. Unfortunately those few bytes contains a +IPD message that the receiving MCU wont see and the communication stalls.

My implementation is the following:
[image]
I have only added mbedtls_chacha20_update() in at_port_write_data() callback to encrypt the data into another buffer before sending it to uart_write_bytes(). Im doing this because if i encrypt the data buffer directly esp-at crashes.

Here is the data from the data buffer before encryption:
[image]

And here is the encrypted data (decrypted again so we can see whats sent from esp-at):
[image]

I have tried to use RC4 instead of chacha20 but i get the same behavior. It seem like the encrypted data gets corupted by the esp-at some how. Is it possible that there is a memory leak somwhere in the esp-at fw that messes up ap_port_write_data() stack?

Im using ESP32-WROOM-32U with 16 MB and version 2.1.0 of esp-at. I have also tried v3.0.0.0 but it wont work either.

Thank you.

ustccw, 2023-04-17:
@LonesomeUnixCowboy Here are some possible troubleshooting directions:

1. Using `uint8_t out[len]` is not usually a good approach, as it may cause stack overflow if len is too large.
2. Check the return value of `mbedtls_chacha20_update()` and compare it with len to confirm if they are the same.
3. If the issue can be easily reproduced, you can print the data before it is encrypted, then print the data after it is decrypted, and separately test this data with `mbedtls_chacha20_update()`.

LonesomeUnixCowboy, 2023-05-02:
Thank you for feedback @ustccw. I have found a solution that seems to work. I have encapsulated ap_port_data_write callback with semaphores and this keeps the encryption to work as expected. So it seems to me that if ap_port_data_write takes so long time it gets interrupted by another task and that causes the corrupted encryption in out[].

closed 2023-05-02 by LonesomeUnixCowboy