espressif/esp-at #704: ESP8266-S1版本编译

MrYanLK, 2023-04-19:
您好，我现在有一个这样的需求，使用ESP8266-S1模块，模块上有一个LED灯用于判断WiFi连接状态，这个LED连接模块上的GPIO4，低电平点亮，高电平熄灭。但是现在我不知道应该怎么在AT示例代码(esp-at/main/interface/uart/at_uart_task.c)中去增加控灯的代码，我想在模块连接时，通过AT去设置GPIO4输出低电平以使灯亮，模块断开连接时，AT设置GPIO4输出高电平以使灯灭。能否帮忙在示例中添加好代码并编译好AT固件给我？

xcguang, 2023-04-19:
可以 MCU 做，MCU 知道Wi-Fi 断开和连接的事件，然后通过命令控制 IO 的状态。这样通用固件就可以满足

MrYanLK, 2023-04-19:
但是那个LED是在esp8266模块上的，不是在MCU上

xcguang, 2023-04-19:
MCU 可以通过 AT+SYSREG 设置对应的 IO 寄存器(具体寄存器可以查技术手册)

MrYanLK, 2023-04-19:
另外，能否发一份ESP8266-S1通用固件给我，我在官网上下载的没法使用，烧录到板子上总是提示ERROR

MrYanLK, 2023-04-19:
您这里说的寄存器技术手册是模块的还是MCU的技术手册呢？

xcguang, 2023-04-20:
ESP8266-S1 不是乐鑫官方模组，请向对应厂商获取固件

xcguang, 2023-04-20:
技术手册，请自行去 https://www.espressif.com/zh-hans/support/documents/technical-documents 下载

MrYanLK, 2023-04-20:
“”MCU 可以通过 AT+SYSREG 设置对应的 IO 寄存器“”---------您的意思是在模块端发送AT+SYSREG指令去设置模块上LED连接的GPIO4吗，但是有个问题，如何判断模块已经连接WiFi了呢？能具体说说流程吗？谢谢

xcguang, 2023-04-20:
模块连接上 Wi-Fi，MCU 会收到消息，请 先将 AT 文档熟悉一下，特别是 https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/index.html#id4

MrYanLK, 2023-04-21:
我这边开了透传后是一套新的通讯协议，不会在进入at指令

MrYanLK, 2023-04-21:
@xcguang

MrYanLK, 2023-04-21:
@xcguang 还有其他建议吗