espressif/esp-at #705: esp-at编译配置

MrYanLK, 2023-04-21:
请问这是什么情况？
[image]

ustccw, 2023-04-21:
根据 https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html 文档看下呢， 似乎是第三步环境没有安装好。

MrYanLK, 2023-04-21:
明明有安装pip模块，为啥还提示这个
[image]

MrYanLK, 2023-04-21:
哦我知道了，是没有运行esp-idf下的. ./export.sh进行配置，现在可以了，但是我想编译一版8266的AT固件，我现在执行./build.py install提示如下：没有esp8266的选项？
[image]

MrYanLK, 2023-04-21:
是需要git checkout release/v2.2.0.0_esp8266 切换分支吧？但是我看乐鑫官网已经有v2.2.1版本了，是否分支也要切换到v2.2.1
[image]

MrYanLK, 2023-04-21:
请帮忙看一下这是什么问题，编译失败了

root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-at#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-at# git checkout release/v2.2.0.0_esp8266
Already on 'release/v2.2.0.0_esp8266'
Your branch is up to date with 'origin/release/v2.2.0.0_esp8266'.
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-at# ./build.py menuconfig
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP8266
3. PLATFORM_ESP32S2
4. PLATFORM_ESP32C3
choose(range[1,4]):2

Module name:
1. WROOM-02 (description: TX:15 RX:13)
2. WROOM-5V2L (description: 5V UART level)
3. ESP8266_1MB (description: No OTA)
4. WROOM-02-N (description: TX:1 RX:3)
5. WROOM-S2
6. ESP8266_QCLOUD (description: QCLOUD TX:15 RX:13)
choose(range[1,6]):1

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):0
platform_name=ESP8266,module_name=WROOM-02
Redirect IDF url to https://gitee.com/EspressifSystems
Please wait for the SDK download to finish...
Cloning into 'esp-idf'...
remote: Enumerating objects: 29431, done.
remote: Counting objects: 100% (199/199), done.
remote: Compressing objects: 100% (196/196), done.
remote: Total 29431 (delta 99), reused 0 (delta 0), pack-reused 29232
Receiving objects: 100% (29431/29431), 63.48 MiB | 1.65 MiB/s, done.
Resolving deltas: 100% (17155/17155), done.
Updating files: 100% (2882/2882), done.
IDF is Cloned from https://gitee.com/EspressifSystems
Submodule 'components/coap/libcoap' (https://gitee.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
Submodule 'components/json/cJSON' (https://gitee.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Submodule 'components/lwip/lwip' (https://gitee.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://gitee.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Submodule 'components/mqtt/esp-mqtt' (https://gitee.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
Redirect submodule.components/json/cJSON.url to https://gitee.com/esp-submodules/cJSON.git
Redirect submodule.components/mbedtls/mbedtls.url to https://gitee.com/esp-submodules/mbedtls.git
Redirect submodule.components/lwip/lwip.url to https://gitee.com/esp-submodules/esp-lwip.git
Redirect submodule.components/mqtt/esp-mqtt.url to https://gitee.com/esp-submodules/esp-mqtt.git
Redirect submodule.components/coap/libcoap.url to https://gitee.com/esp-submodules/libcoap.git
Update submodule...
Cloning into '/home/ylk/esp/esp-at/esp-idf/components/coap/libcoap'...
Cloning into '/home/ylk/esp/esp-at/esp-idf/components/json/cJSON'...
Cloning into '/home/ylk/esp/esp-at/esp-idf/components/lwip/lwip'...
Cloning into '/home/ylk/esp/esp-at/esp-idf/components/mbedtls/mbedtls'...
Cloning into '/home/ylk/esp/esp-at/esp-idf/components/mqtt/esp-mqtt'...
Submodule path 'components/coap/libcoap': checked out 'cfec0d072c5b99ed3e54828ca50ea2f6b91e1f50'
Submodule path 'components/json/cJSON': checked out '3c8935676a97c7c97bf006db8312875b4f292f6c'
Submodule path 'components/lwip/lwip': checked out '4fd84abace0cd8a5a709c194fa54afb3cd39b3f7'
Submodule path 'components/mbedtls/mbedtls': checked out '9ef92c551eb8d92677034c3ec8078a8076febf41'
Submodule path 'components/mqtt/esp-mqtt': checked out '01594bf118ae502b5a0ead040446f2be75d26223'
old commit:bac80c938c109561ae408245073fb6d7694fb5a3
checkout commit:e3348ac7eaf1d91d73a9de1eda2c430ef1afe4bc
Please wait for the update to complete, which will take some time
Already up to date.
Note: switching to 'e3348ac7eaf1d91d73a9de1eda2c430ef1afe4bc'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

[... 202 lines cut ...]
  CMakeLists.txt:53 (include)

CMake Warning at /home/ylk/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule
  examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib
  is out of date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/ylk/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/ylk/esp-idf/tools/cmake/build.cmake:175 (__build_get_idf_git_revision)
  esp-idf/tools/cmake/idf.cmake:43 (__build_init)
  esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:53 (include)

CMake Error at /home/ylk/esp-idf/tools/cmake/targets.cmake:81 (message):
  Toolchain file /home/ylk/esp-idf/tools/cmake/toolchain-esp8266.cmake not
  found
Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:245 (__target_set_toolchain)
  CMakeLists.txt:55 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
idf.py build ret: 2
idf.py build failed
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-at#

MrYanLK, 2023-04-23:
Hi ，我这边使用esp8266-s1模块，电脑端使用tcp调试助手做TCP server，模块连接server，服务端定时下发心跳，模块端收到消息后需要做出应答。使用乐鑫官网下载的最新固件没有问题，但是使用自己编译的AT固件就会出现不稳定，模块长时间接收不到心跳或者突然接收到一大堆心跳消息等，编译的固件和官网下载的都是v2.2.1版本，只是自己编译的改了个串口而已，将烧录和发AT都设置在uart0了，其他没动过，请问下这可能是什么原因造成的不稳定呢？能否原厂帮忙编译一下，配置烧录和发AT都在同口uart0
[image]
[image]

MrYanLK, 2023-04-23:
能否帮忙看看，此项目较为紧急

MrYanLK, 2023-04-24:
我在运行esp-at目录下的esp-idf/export.sh时报如下错误
[image]

MrYanLK, 2023-04-25:
非常感谢，目前我测了一下这个固件，烧录和发AT是同口的，请问这个固件还要其他区别或者改动吗？能否给到客户作为量产固件呢？

xcguang, 2023-04-25:
软件没有什么变动，如果提供给客户，请自行备份好固件，并根据产品做功能验证

MrYanLK, 2023-04-25:
您好，客户反馈使用这个固件还是会出现通信不稳定，在13:11:47后又有8秒没接收到数据，请帮忙看下
[image]
使用下面红框中的就比较稳定，可惜这个固件烧录和发AT不是同串口，客户板子上没有预留发AT的IO13和IO15
[image]

MrYanLK, 2023-04-25:
模块内是不是带了某个操作系统，是不是操作系统的调度问题？