espressif/esp-at #706: [at-mqtt]error code:0x6006

mengxiaohe, 2023-04-22:
[at-mqtt]error code:0x6006

MQTT CONNECTED

[0;31mE (413028) MQTT_CLIENT: Writing didn't complete in specified timeout: errno=0[0m
MQTT ERROR

MQTT DISCONNECTED

[at-mqtt]error code:0x600C

at固件mqtt 发多了之后无法发送,但是在mqtt服务端上是能够看到客户端在线的

mengxiaohe, 2023-04-22:
型号为ESP32 C3 V2.4.2

ustccw, 2023-05-04:
从日志看，是在发布一条 MQTT 消息时，超时失败了。可以打开设备端日志以及配合抓包进一步排查是哪里的问题。
设备端日志使能：
- System: `python build.py menuconfig` -> `Component config` -> `Log output` -> `Default log verbosity` -> `Debug`
- mbedTLS: `python build.py menuconfig` -> `Component config` -> `mbedTLS` -> `Enable mbedTLS debuging` -> `Set mbedTLS debugging level` -> `Verbose`
- LwIP(TCP): `python build.py menuconfig` -> `Component config` -> `LWIP` -> `Enable LWIP Debug` -> `<Enable IP, TCP, socket related choices`

closed 2023-07-17 by ustccw