espressif/esp-at #707: esp-at分支问题

MrYanLK, 2023-04-24:
您好，我在执行./build.py menuconfig时报了这个问题，请问下怎么解决？
[image]

MrYanLK, 2023-04-24:
[image]

ustccw, 2023-05-04:
将目录下的 `build sdkconfig esp-idf` 删除掉，重新 `./build.py install` 和 `./build.py menuconfig` 试试。

closed 2023-05-18 by MrYanLK

MrYanLK, 2023-05-23:
好像是因为我先编译过ESP32 的AT固件，当前修改过文件，现在切换到8266的分支上就提示这个，我对git不太熟，不知怎么提交修改或丢弃修改，

error: Your local changes to the following files would be overwritten by checkout:
        components/customized_partitions/raw_data/factory_param/factory_param_data.csv
Please commit your changes or stash them before you switch branches.
Aborting