espressif/esp-at #711: OTA update using command AT+CIUPDATE unclear - iot.espressif.cn used but is not operational

fahirad, 2023-05-03:
The official documentation seems to not be up to date. 

[https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_implement_OTA_update.html#upgrade-commands-use](url)

It states that the server used for updating the official firmware is iot.espressif.cn (for the usage of AT+CIUPDATE command), however this server doesn't seem to be operational. The latest version also has it as the default server in the menuconfig, so I suppose this needs to be changed or updated in some way.

ustccw, 2023-05-04:
@fahirad Thank you very much for your suggestion. This document is indeed a bit outdated and will be updated. Thank you!

ustccw, 2023-06-09:
@fahirad I'm afraid this document won't be moved out, since there is a note in step 2:

**The Join function is currently not open to new users. If you want to use it, please contact [Espressif](https://www.espressif.com/en/contact-us/sales-questions).**

In other words, if your product does require this feature, please contact the business to obtain an account. Once you have an account, this document will guide you to continue developing your product. Please know.

~