espressif/esp-at #713: 好,想要分段下载,怎么操作呢?

hzhh110, 2023-05-06:
你好,我这边能否分段下载呢?

busy p...

OK
AT+GMR

busy p...
AT version:2.4.1.0(dbcd258 - ESP32C3 - Aug  4 2022 06:58:36)
SDK version:qa-test-v4.3.3-20220423
compile time(205b068):Aug  9 2022 09:07:01
Bin version:2.4.1(MINI-1)

OK
AT+HTTPCHEAD?

busy p...

ERROR
AT+HTTPCHEAD?

busy p...

ERROR

Range:bytes=0-100

ERROR

AT+HTTPCGET="https://juhuico.oss-cn-hangzhou.aliyuncs.com/versionInfo/20230415/iDoor_150943_nq3j.0.1.bin?Expires=4803606583&OSSAccessKeyId=LTAI7b3shRx5MYws&Signature=K7n%2Bw7GpeyeXGZJscf5Lw5iXsag%3D"

发现设置请求头失败,然后有哪个方式设置Range呢?

hzhh110, 2023-05-06:
我想要下载部分后,存储成功了后,再下载下一段

ustccw, 2023-05-06:
@hzhh110 `AT+HTTPCHEAD?` 是最近添加的命令。你的固件是 v2.4.1 版本的，可以试试使用较新的版本测试验证。

hzhh110, 2023-05-08:
> @hzhh110 `AT+HTTPCHEAD?` 是最近添加的命令。你的固件是 v2.4.1 版本的，可以试试使用较新的版本测试验证。

确实,最新版本确实有该功能,想了解下,有没有更高效的方法做分段下载呢?比如先下载到ESP32的文件缓存,然后通过分段读取文件缓存再打印出来,这样效果会比直接设置range再请求服务器文件效率高

hzhh110, 2023-05-08:
> @hzhh110 `AT+HTTPCHEAD?` 是最近添加的命令。你的固件是 v2.4.1 版本的，可以试试使用较新的版本测试验证。

确实,最新版本确实有该功能,想了解下,有没有更高效的方法做分段下载呢?这个确实也可以实现,但当文件800k,要是分成每4K一次RANGE,那样效率很低,或者有没有好点的建议呢?

ustccw, 2023-05-09:
一定要分段下载吗。如果 MCU 能够处理的过来，就直接一次性下载。

closed 2023-07-17 by ustccw