espressif/esp-at #717: Problem with AT driver commands

JakaKor, 2023-05-12:
Hello all, 

I want to use the ESP32-WROOM as a bridge between my main MCU and an I2C sensor. Since I don't have spare pins on the main MCU, I intend to control the I2C interface on the ESP32 using AT+DRV* commands. However, I'm encountering difficulties in making it work. Whenever I send the command 'AT+DRVI2CINIT=0,19,18,100000\r\n,' the response I receive is always 'ERROR.'

I have enabled the AT driver commands in the menuconfig.

A snippet from serial monitor:
```
AT+DRVI2CINIT=0,19,18,100000

ERROR
AT+GMR
AT version:2.4.0.0(s-4c6eb5e - ESP32 - May 20 2022 03:12:58)
SDK version:v4.3.5-83-g33b3db35d7
compile time(5641e0a):May 11 2023 10:15:33
Bin version:2.4.0(WROOM-32)

OK
```

Thank you for your help.

ustccw, 2023-05-15:
@JakaKor From the version, it seems that you are using an incompatible ESP-IDF version (`33b3db35d7`). In the ESP-AT project, modifying the ESP-IDF version often leads to various unknown issues. It is recommended to use the ESP-IDF version specified in AT, which can be found in `module_config/module_esp32_default/IDF_VERSION`.

Here is the result of my local testing: 
[image]

JakaKor, 2023-05-15:
Thank you for your answer. 

I must be doing something wrong, because I had some problems with the compilation of the project. 
First I tried with the latest version. I compiled and flash the project successfully, but I couldn't get the BLE services to start. 

```
AT+GMR
AT version:3.1.0.0-dev(s-bd890df - ESP32 - May 12 2023 08:47:33)
SDK version:v5.0-541-g885e501d99
compile time(c5af3bc):May 15 2023 12:00:57
Bin version:2.5.0(WROOM-32)

OK
```

When I try to set up the BLE I get the following:

```
AT+BLEINIT=2

busy p...

OK

AT+BLEGATTSSRVCRE

busy p...

OK

AT+BLEGATTSSRV?

busy p...
+BLEGATTSSRV:1,0,0x,1
+BLEGATTSSRV:2,0,0x,1

OK

```

I can connect to ESP with my phone, but there isn't any service or characteristics visible.
I think I get similar problem as described in issue #696

My solution was that I checkout the release for 2.4 ESP AT (5641e0aa2b3cbc36b2d843a72dd00d248daaef40), but I was unable to compile it with the ESP-IDF 5.0. 
Probably because of the different versions?

```
branch:release/v4.3
commit:588436db47da792736dade1c21fcbf2c879456a4
repository:https://github.com/espressif/esp-idf.git
```

```
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.0_py3.11_env
    Python interpreter used: C:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe

ESP-IDF v4.3.2-553-g588436db47
A fatal error occurred: idf.py build failed
```

I tried to compile it with ESP-IDF 4.3, without changing the branch I get similar error:

```
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf4.3_py3.11_env
    Python interpreter used: C:\Espressif\python_env\idf4.3_py3.11_env\Scripts\python.exe

ESP-IDF v4.3.2-553-g588436db47
A fatal error occurred: idf.py build failed
```

Later I changed the commit to (33b3db35d73d60d12dae52157b2f5c5dbe1a813c)

build and flash are successful.

serial monitor snippet:

```
AT+GMR
AT version:2.4.1.0(s-fb731d2 - ESP32 - Aug 25 2022 03:48:17)
SDK version:v4.3.5-83-g33b3db35d7
compile time(f64aa35):May 15 2023 13:19:23
Bin version:2.4.1(WROOM-32)

OK
AT+BLEGATTSSRVCRE

busy p...

OK
AT+BLEGATTSSRV?

busy p...
+BLEGATTSSRV:1,0,0xA002,1
+BLEGATTSSRV:2,0,0xA003,1

OK
AT+BLEGATTSSRVSTART=1

busy p...

OK
AT+BLESECPARAM=1,3,16,3,3,0

busy p...

OK
AT+DRVI2CINIT=0,19,18,100000

ERROR
```

That is why I decided to use this version, because I couldn't make ESP working in other communication. 

Thank you for your help.

ustccw, 2023-05-22:
@JakaKor could you please create a new issue about BLE service, i would like to ask my colleagues to help with this issue.

ustccw, 2023-06-05:
We assume this worked for you, and will close it. About BLE issue, move to https://github.com/espressif/esp-at/issues/725, and wait for BLE guys to fix.

closed 2023-06-05 by ustccw