espressif/esp-at #718: Can't set UART speed esp32 c3 mini

JakaKor, 2023-05-16:
Hello all, 

I have a problem setting different baudrate than 115200. 
I tried to change the factory_param_data.csv.

```
platform,module_name,description,version,max_tx_power,uart_port,start_channel,channel_num,country_code,uart_baudrate,uart_tx_pin,uart_rx_pin,uart_cts_pin,uart_rts_pin
PLATFORM_ESP32,WROOM-32,,4,78,1,1,13,CN,115200,17,16,15,14
PLATFORM_ESP32,WROVER-32,,4,78,1,1,13,CN,115200,22,19,15,14
PLATFORM_ESP32,PICO-D4,,4,78,1,1,13,CN,115200,22,19,15,14
PLATFORM_ESP32,SOLO-1,,4,78,1,1,13,CN,115200,17,16,15,14
PLATFORM_ESP32,MINI-1,ESP32-U4WDH chip inside,4,78,1,1,13,CN,115200,22,19,15,14
PLATFORM_ESP32,ESP32-SDIO,,4,78,-1,1,13,CN,-1,-1,-1,-1,-1
PLATFORM_ESP32,ESP32-D2WD,"2MB flash, No OTA",4,72,1,1,13,CN,115200,22,19,15,14
**PLATFORM_ESP32C3,MINI-1,TX:77 RX:66,4,72,1,1,13,SI,57600,7,6,5,4**
PLATFORM_ESP32C3,ESP32C3-SPI,,4,78,-1,1,13,CN,-1,-1,-1,-1,-1
PLATFORM_ESP32C3,ESP32C3_RAINMAKER,TX:7 RX:6,4,78,1,1,13,CN,115200,7,6,5,4
PLATFORM_ESP32C2,ESP32C2-2MB,"ESP32C2 (ESP8684) series with 2MB flash, TX:7 RX:6",4,78,1,1,13,CN,115200,7,6,19,20
PLATFORM_ESP32C2,ESP32C2-4MB,"ESP32C2 (ESP8684) series with 4MB flash, TX:7 RX:6",4,78,1,1,13,CN,115200,7,6,5,4
```

I tried to change the description "TX:77 RX:66" just to see if it changes anything, and it works as expected.

[image]

I changed max tx power to 72, and also this changed. But as you can see on the picture, the uart baud rate didn't change

[image]

thank you for your help.

ustccw, 2023-05-17:
@JakaKor Hello, it looks like the baud rate 115200 was used before. Please try erasing the flash by running the command `./build.py erase_flash` before flashing the firmware and test again.
The current logic of AT is as follows: it first checks if the UART configuration is saved in NVS. If it exists, it uses that configuration. If not, it uses the configuration defined in `esp-at/components/customized_partitions/raw_data/factory_param/factory_param_data.csv` and saves it to NVS.

~

JakaKor, 2023-05-17:
Thank you. 

Erasing flash solved the problem, and I am able to change the baud rate.

closed 2023-05-17 by JakaKor