espressif/esp-at #721: esp-at编译配置失败

MrYanLK, 2023-05-17:
您好，esp-idf已能正常编译项目，esp-at/esp-idf下也正常执行了./install 和 . ./export.sh ，但是我在esp-at目录下执行./build.py menuconfig时报错如下

root@DESKTOP-6TBOMGV:/home/ylk/esp-at# ./build.py menuconfig
Platform name:ESP32     Module name:WROOM-32    Silence:0
submodules check completed for updates.
softap_pmk.patch does not need to be applied
patches check completed for updates.
Ready to set up environment variables..
PATH is /home/ylk/esp-at/esp-idf/components/esptool_py/esptool:/home/ylk/esp-at/esp-idf/components/espcoredump:/home/ylk/esp-at/esp-idf/components/partition_table:/home/ylk/esp-at/esp-idf/components/app_update:/home/ylk/esp-at/esp-idf/components/esptool_py/esptool:/home/ylk/esp-at/esp-idf/components/espcoredump:/home/ylk/esp-at/esp-idf/components/partition_table:/home/ylk/esp-at/esp-idf/components/app_update:/root/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/root/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/root/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/root/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/root/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/root/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/root/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/root/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/root/.espressif/python_env/idf5.0_py3.10_env/bin:/home/ylk/esp-at/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is /root/.espressif/python_env/idf5.0_py3.10_env
sys.platform is linux
export str is
PATH is /home/ylk/esp-at/esp-idf/components/esptool_py/esptool:/home/ylk/esp-at/esp-idf/components/espcoredump:/home/ylk/esp-at/esp-idf/components/partition_table:/home/ylk/esp-at/esp-idf/components/app_update:/home/ylk/esp-at/esp-idf/components/esptool_py/esptool:/home/ylk/esp-at/esp-idf/components/espcoredump:/home/ylk/esp-at/esp-idf/components/partition_table:/home/ylk/esp-at/esp-idf/components/app_update:/root/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/root/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/root/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/root/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/root/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/root/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/root/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/root/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/root/.espressif/python_env/idf5.0_py3.10_env/bin:/home/ylk/esp-at/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is /root/.espressif/python_env/idf5.0_py3.10_env
Executing action: menuconfig
Running cmake in directory /home/ylk/esp-at/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/ylk/esp-at"...
silence:0 sdkconfig:/home/ylk/esp-at/module_config/module_esp32_default/sdkconfig.defaults
esp-at branch: master
esp-at docs version: latest
-- Found Git: /usr/bin/git (found version "2.34.1")
CMake Warning at esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule
  components/bootloader/subproject/components/micro-ecc/micro-ecc is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:89 (include)

CMake Warning at esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32 is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:89 (include)

CMake Warning at esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c2/esp32c2-bt-lib is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:89 (include)

CMake Warning at esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c3_family is out of date.
  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:89 (include)

CMake Warning at esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32h2/esp32h2-bt-lib is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  esp-idf/tools/cmake/project.cmake:12 (include)
[... 187 lines cut ...]
      data = self._fp_read(amt) if not fp_closed else b""
    File "/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/urllib3/response.py", line 794, in _fp_read
      return self._fp.read(amt) if amt is not None else self._fp.read()
    File "/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/filewrapper.py", line 96, in read
      self._close()
    File "/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/filewrapper.py", line 76, in _close
      self.__callback(result)
    File "/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/controller.py", line 353, in cache_response
      self._cache_set(cache_url, request, response, body, expires_time)
    File "/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/controller.py", line 274, in _cache_set
      self.serializer.dumps(request, response, body),
    File "/root/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/serialize.py", line 54, in dumps
      u"strict": response.strict,

  AttributeError: 'HTTPResponse' object has no attribute 'strict'

Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:91 (project)

-- Configuring incomplete, errors occurred!
See also "/home/ylk/esp-at/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
A fatal error occurred: idf.py build failed
root@DESKTOP-6TBOMGV:/home/ylk/esp-at#

xcguang, 2023-05-17:
python -m pip install urllib3==1.26.0 试试

MrYanLK, 2023-05-17:
好像可以了，感谢！