espressif/esp-at #723: AT固件编译

MrYanLK, 2023-05-18:
你好，我再执行./build.py build时报错如下，请问是咋回事

yyy@DESKTOP-6TBOMGV:~/esp-at$ ./build.py build
Platform name:ESP32C3   Module name:MINI-1      Silence:0
submodules check completed for updates.
softap_pmk.patch does not need to be applied
patches check completed for updates.
Ready to set up environment variables..
PATH is /home/yyy/esp-at/esp-idf/components/esptool_py/esptool:/home/yyy/esp-at/esp-idf/components/espcoredump:/home/yyy/esp-at/esp-idf/components/partition_table:/home/yyy/esp-at/esp-idf/components/app_update:/home/yyy/esp-at/esp-idf/components/esptool_py/esptool:/home/yyy/esp-at/esp-idf/components/espcoredump:/home/yyy/esp-at/esp-idf/components/partition_table:/home/yyy/esp-at/esp-idf/components/app_update:/home/yyy/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/yyy/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/yyy/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/yyy/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/yyy/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/yyy/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/yyy/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/yyy/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/yyy/.espressif/python_env/idf5.0_py3.8_env/bin:/home/yyy/esp-at/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/d/VSCode/Microsoft VS Code/bin:/mnt/d/git/Git/cmd:/mnt/c/Program Files/usbipd-win/:/mnt/d/Mingw_win32_w64/x86_64-8.1.0-release-win32-seh-rt_v6-rev0/mingw64/bin:/snap/bin
IDF_PYTHON_ENV_PATH is /home/yyy/.espressif/python_env/idf5.0_py3.8_env
sys.platform is linux
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
export str is
PATH is /home/yyy/esp-at/esp-idf/components/esptool_py/esptool:/home/yyy/esp-at/esp-idf/components/espcoredump:/home/yyy/esp-at/esp-idf/components/partition_table:/home/yyy/esp-at/esp-idf/components/app_update:/home/yyy/esp-at/esp-idf/components/esptool_py/esptool:/home/yyy/esp-at/esp-idf/components/espcoredump:/home/yyy/esp-at/esp-idf/components/partition_table:/home/yyy/esp-at/esp-idf/components/app_update:/home/yyy/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/yyy/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/yyy/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/yyy/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/yyy/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/yyy/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/yyy/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/yyy/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/yyy/.espressif/python_env/idf5.0_py3.8_env/bin:/home/yyy/esp-at/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/d/VSCode/Microsoft VS Code/bin:/mnt/d/git/Git/cmd:/mnt/c/Program Files/usbipd-win/:/mnt/d/Mingw_win32_w64/x86_64-8.1.0-release-win32-seh-rt_v6-rev0/mingw64/bin:/snap/bin
IDF_PYTHON_ENV_PATH is /home/yyy/.espressif/python_env/idf5.0_py3.8_env
Executing action: all (aliases: build)
Running ninja in directory /home/yyy/esp-at/build
Executing "ninja all"...
silence:0 sdkconfig:/home/yyy/esp-at/module_config/module_esp32c3_default/sdkconfig.defaults
esp-at branch: master
esp-at docs version: latest
[0/1] Re-running CMake...-- Component directory /home/yyy/esp-at/components/fs_image does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/yyy/esp-at/bootloader_components/include does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/yyy/esp-at/bootloader_components/tools does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32c3
Processing 3 dependencies:
[3/3] idf (5.0.0)-- Project sdkconfig file /home/yyy/esp-at/sdkconfig
Loading defaults file /home/yyy/esp-at/module_config/module_esp32c3_default/sdkconfig.defaults...
-- App "esp-at" version: v3.0.0.0-138-gc5af3bc
-- Adding linker script /home/yyy/esp-at/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/yyy/esp-at/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script /home/yyy/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /home/yyy/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /home/yyy/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /home/yyy/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /home/yyy/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script /home/yyy/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script /home/yyy/esp-at/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
silence:0, ld core:/home/yyy/esp-at/components/at/lib/libesp32c3_at_core.a
Parsing CSV input...
Verifying table...
-- Components: app_trace app_update at bootloader bootloader_support bt cmock console customized_partitions cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp_websocket_client espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/yyy/esp-at/esp-idf/components/app_trace /home/yyy/esp-at/esp-idf/components/app_update /home/yyy/esp-at/components/at /home/yyy/esp-at/esp-idf/components/bootloader /home/yyy/esp-at/esp-idf/components/bootloader_support /home/yyy/esp-at/esp-idf/components/bt /home/yyy/esp-at/esp-idf/components/cmock /home/yyy/esp-at/esp-idf/components/console /home/yyy/esp-at/components/customized_partitions /home/yyy/esp-at/esp-idf/components/cxx /home/yyy/esp-at/esp-idf/components/driver /home/yyy/esp-at/esp-idf/components/efuse /home/yyy/esp-at/esp-idf/components/esp-tls /home/yyy/esp-at/esp-idf/components/esp_adc /home/yyy/esp-at/esp-idf/components/esp_app_format /home/yyy/esp-at/esp-idf/components/esp_common /home/yyy/esp-at/esp-idf/components/esp_eth /home/yyy/esp-at/esp-idf/components/esp_event /home/yyy/esp-at/esp-idf/components/esp_gdbstub /home/yyy/esp-at/esp-idf/components/esp_hid /home/yyy/esp-at/esp-idf/components/esp_http_client /home/yyy/esp-at/esp-idf/components/esp_http_server /home/yyy/esp-at/esp-idf/components/esp_https_ota /home/yyy/esp-at/esp-idf/components/esp_https_server /home/yyy/esp-at/esp-idf/components/esp_hw_support /home/yyy/esp-at/esp-idf/components/esp_lcd /home/yyy/esp-at/esp-idf/components/esp_local_ctrl /home/yyy/esp-at/esp-idf/components/esp_netif /home/yyy/esp-at/esp-idf/components/esp_partition /home/yyy/esp-at/esp-idf/components/esp_phy /home/yyy/esp-at/esp-idf/components/esp_pm /home/yyy/esp-at/esp-idf/components/esp_psram /home/yyy/esp-at/esp-idf/components/esp_ringbuf /home/yyy/esp-at/esp-idf/components/esp_rom /home/yyy/esp-at/esp-idf/components/esp_system /home/yyy/esp-at/esp-idf/components/esp_timer /home/yyy/esp-at/esp-idf/components/esp_wifi /home/yyy/esp-at/esp-idf/components/espcoredump /home/yyy/esp-at/managed_components/espressif__esp_websocket_client /home/yyy/esp-at/managed_components/espressif__mdns /home/yyy/esp-at/esp-idf/components/esptool_py /home/yyy/esp-at/esp-idf/components/fatfs /home/yyy/esp-at/esp-idf/components/freertos /home/yyy/esp-at/esp-idf/components/hal /home/yyy/esp-at/esp-idf/components/heap /home/yyy/esp-at/esp-idf/components/http_parser /home/yyy/esp-at/esp-idf/components/idf_test /home/yyy/esp-at/esp-idf/components/ieee802154 /home/yyy/esp-at/esp-idf/components/json /home/yyy/esp-at/esp-idf/components/log /home/yyy/esp-at/esp-idf/components [... cut ...]
0x8000,partition_table/partition-table.bin
0xd000,ota_data_initial.bin
0xf000,phy_multiple_init_data.bin
0x1e000,at_customize.bin
0x1f000,customized_partitions/mfg_nvs.bin
Create /home/yyy/esp-at/build/factory/factory_MINI-1.bin for MINI-1 finished
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/yyy/esp-at/build/log/idf_py_stderr_output_22309 and /home/yyy/esp-at/build/log/idf_py_stdout_output_22309
A fatal error occurred: idf.py build failed
yyy@DESKTOP-6TBOMGV:~/esp-at$

MrYanLK, 2023-05-18:
另外，我在menuconfig中打开blufi相关的配置，编译时报错如下

root@DESKTOP-6TBOMGV:/home/ylk/esp-at# ./build.py build
Platform name:ESP32C2   Module name:ESP32C2-2MB Silence:0
submodules check completed for updates.
blufi-adv.patch does not need to be applied
wifi_deinit.patch does not need to be applied
softap_pmk.patch does not need to be applied
patches check completed for updates.
Ready to set up environment variables..
PATH is /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is None
sys.platform is linux
export str is OPENOCD_SCRIPTS=/root/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
ESP_ROM_ELF_DIR=/root/.espressif/tools/esp-rom-elfs/20220823/
IDF_PYTHON_ENV_PATH=/root/.espressif/python_env/idf5.0_py3.10_env
ESP_IDF_VERSION=5.0
PATH=/root/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/root/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/root/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/root/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/root/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/root/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/root/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/root/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/root/.espressif/python_env/idf5.0_py3.10_env/bin:/home/ylk/esp-at/esp-idf/tools:$PATH
IDF_DEACTIVATE_FILE_PATH=/tmp/tmptrwl4kididf_1525

PATH is /root/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/root/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/root/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/root/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/root/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/root/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/root/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/root/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/root/.espressif/python_env/idf5.0_py3.10_env/bin:/home/ylk/esp-at/esp-idf/tools:$PATH:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is /root/.espressif/python_env/idf5.0_py3.10_env
Executing action: all (aliases: build)
Running ninja in directory /home/ylk/esp-at/build
Executing "ninja all"...
silence:0 sdkconfig:/home/ylk/esp-at/module_config/module_esp32c2-2mb/sdkconfig.defaults
esp-at branch: master
esp-at docs version: latest
[0/1] Re-running CMake...-- Component directory /home/ylk/esp-at/components/fs_image does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ylk/esp-at/bootloader_components/include does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ylk/esp-at/bootloader_components/tools does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32c2
Processing 3 dependencies:
[3/3] idf (5.0.0)-- Project sdkconfig file /home/ylk/esp-at/sdkconfig
Loading defaults file /home/ylk/esp-at/module_config/module_esp32c2-2mb/sdkconfig.defaults...
-- App "esp-at" version: v3.0.0.0-136-g14bad8d
-- Adding linker script /home/ylk/esp-at/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_system/ld/esp32c2/sections.ld.in
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.api.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.rvfp.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.newlib.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.version.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.mbedtls.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.newlib-time.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.newlib-nano.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.heap.ld
-- Adding linker script /home/ylk/esp-at/esp-idf/components/soc/esp32c2/ld/esp32c2.peripherals.ld
silence:0, ld core:/home/ylk/esp-at/components/at/lib/libesp32c2_at_core.a
Parsing CSV input...
Verifying table...
-- Components: app_trace app_update at bootloader bootloader_support bt cmock console customized_partitions cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp_websocket_client espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
[... cut ...]
/root/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld: (.text.at_setupCmdBleInit+0xac): undefined reference to `ble_gap_periodic_adv_sync_terminate'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ylk/esp-at/build/log/idf_py_stderr_output_1600 and /home/ylk/esp-at/build/log/idf_py_stdout_output_1600
A fatal error occurred: idf.py build failed

ustccw, 2023-05-22:
@MrYanLK 第一个错误是：在编译 ESP32-C3 的 MINI-1 模组的固件，但是环境是 ESP32-C2 的。你可以试试将 build sdkconfig bootloader_components 都删除掉，重新编译 ESP32-C3 的固件。

第二个错误：在编译 ESP32-C2 的 2MB 模组的固件时，除了开启 https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32c2/AT_Command_Set/BLE_AT_Commands.html#bluetooth-low-energy-at 这两个配置之外，另外需要禁用下面的配置。
- `Component config` -> `Bluetooth` -> `NimBLE Options` -> 取消勾选 `Enable BLE 5 feature` 
- `Component config` -> `AT` ->  取消勾选 `AT ble command support.`

MrYanLK, 2023-05-22:
问题已解决，感谢！

closed 2023-05-22 by ustccw

MrYanLK, 2023-06-06:
你好，我在编译esp-at时遇到新问题，麻烦看一下是怎么回事儿

ylk@ylk-virtual-machine:~/Desktop/espat/esp-at$ ./build.py build
Platform name:ESP32	Module name:WROOM-32	Silence:0
submodules check completed for updates.
softap_pmk.patch does not need to be applied
patches check completed for updates.
Ready to set up environment variables..
PATH is /home/ylk/Desktop/espat/esp-at/esp-idf/components/esptool_py/esptool:/home/ylk/Desktop/espat/esp-at/esp-idf/components/espcoredump:/home/ylk/Desktop/espat/esp-at/esp-idf/components/partition_table:/home/ylk/Desktop/espat/esp-at/esp-idf/components/app_update:/home/ylk/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/ylk/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/ylk/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/ylk/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/ylk/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/ylk/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/ylk/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/ylk/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/ylk/.espressif/python_env/idf5.0_py3.8_env/bin:/home/ylk/Desktop/espat/esp-at/esp-idf/tools:/home/ylk/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is /home/ylk/.espressif/python_env/idf5.0_py3.8_env
sys.platform is linux
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
export str is 
PATH is /home/ylk/Desktop/espat/esp-at/esp-idf/components/esptool_py/esptool:/home/ylk/Desktop/espat/esp-at/esp-idf/components/espcoredump:/home/ylk/Desktop/espat/esp-at/esp-idf/components/partition_table:/home/ylk/Desktop/espat/esp-at/esp-idf/components/app_update:/home/ylk/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/ylk/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/ylk/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/ylk/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/ylk/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/ylk/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/ylk/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/ylk/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/ylk/.espressif/python_env/idf5.0_py3.8_env/bin:/home/ylk/Desktop/espat/esp-at/esp-idf/tools:/home/ylk/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
IDF_PYTHON_ENV_PATH is /home/ylk/.espressif/python_env/idf5.0_py3.8_env
Executing action: all (aliases: build)
Running ninja in directory /home/ylk/Desktop/espat/esp-at/build
Executing "ninja all"...
[1/1] cd /home/ylk/Desktop/espat/esp-at/build/bootloader/esp-idf/esptool_py && /home/ylk/...t 0x8000 bootloader 0x1000 /home/ylk/Desktop/espat/esp-at/build/bootloader/bootloader.binBootloader binary size 0x6880 bytes. 0x780 bytes (7%) free.
[2/6] Running utility command for customized_bin
Creating NVS binary with version: V2 - Multipage Blob Support Enabled

Created NVS binary: ===> /home/ylk/Desktop/espat/esp-at/build/customized_partitions/mfg_nvs.bin
[3/6] Linking CXX executable esp-at.elfFAILED: esp-at.elf 
: && /home/ylk/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address   CMakeFiles/esp-at.elf.dir/project_elf_src_esp32.c.obj  -o esp-at.elf  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp [... cut ...]
ninja: build stopped: subcommand failed.
HINT: The applications static IRAM usage is larger than the available IRAM size.
For more information on how to reduze IRAM usage run 'idf.py docs -sp api-guides/performance/ram-usage.html#optimizing-iram-usage' 
ninja failed with exit code 1, output of the command is in the /home/ylk/Desktop/espat/esp-at/build/log/idf_py_stderr_output_37501 and /home/ylk/Desktop/espat/esp-at/build/log/idf_py_stdout_output_37501
A fatal error occurred: idf.py build failed

MrYanLK, 2023-06-06:
@ustccw