espressif/esp-at #724: tcp send data step

xiaoheng14, 2023-05-20:
please tell me when use `CIPSEND` command have to first send `CIPSEND=4096` and wait for `>`
can we just use `CIPSEND=data` ? beacuse two step to send data takes a little long time, may be it can make it fast to send data with one step.

MaJerle, 2023-05-20:
You always have to wait for `> ` feedback, regardless of command.

xiaoheng14, 2023-05-21:
thank you for your response, but i think just send data with one step could make it faster, will it be support latter?

ustccw, 2023-05-22:
@xiaoheng14 you could consider one of the following ways:
- Send the big data by [AT+CIPSENDL](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#cmd-sendl) command
- Send data in the passthough mode, [Example](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/TCP-IP_AT_Examples.html#uart-wi-fi-passthrough-transmission-when-the-esp32-works-as-a-tcp-client-in-single-connection).
- Re-define `AT+CIPSEND` command according to your product. There is a hidden API called `int32_t esp_at_get_socket_by_link_id(uint8_t id)`, use the returned socket fd send data.

xiaoheng14, 2023-05-22:
thank you for your advise，I will test these methods latter.

xiaoheng14, 2023-05-22:
> @xiaoheng14 you could consider one of the following ways:
>
> * Send the big data by [...]

thank you for your advise，I will test these methods latter.

xiaoheng14, 2023-06-02:
> thank you for your advise，I will test these methods latter.

请问如果是单连接的话是使用哪个接口获取socket fd呢？

ustccw, 2023-06-05:
@xiaoheng14 the same API, and the link_id is 0. you couly try to get the socket fd by `esp_at_get_socket_by_link_id(0)`

xiaoheng14, 2023-06-05:
thank you, but now i have another problem, when i use esp_at_port_read_data api，if i send data not at all once, it cannot recv the remain data.

xiaoheng14, 2023-06-05:
thank you, but now i have another problem, when i use esp_at_port_read_data api，if i send data not at all once, it cannot recv the remain data. like this:  send command AT+TEST=10\r\n; and then i input data: hello, and esp_at_port_read_data is done..., the recv data become "hello\0\0\0\0\0"

xiaoheng14, 2023-06-05:
here is the code from https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32c3/Compile_and_Develop/How_to_add_user-defined_AT_commands.html

uint8_t at_setup_cmd_test(uint8_t para_num)
{
    int32_t specified_len = 0;
    int32_t received_len = 0;
    int32_t remain_len = 0;
    uint8_t *buf = NULL;
    uint8_t buffer[64] = {0};

    if (esp_at_get_para_as_digit(0, &specified_len) != ESP_AT_PARA_PARSE_RESULT_OK) {
        return ESP_AT_RESULT_CODE_ERROR;
    }

    // buf = (uint8_t *)malloc(specified_len);
    buf = (uint8_t *) calloc(1, specified_len);
    if (buf == NULL) {
        memset(buffer, 0, 64);
        snprintf((char *)buffer, 64, "malloc failed\r\n");
        esp_at_port_write_data(buffer, strlen((char *)buffer));
    }

    // 示例代码
    // 不必在此处创建 semaphores
    if (!at_sync_sema) {
        at_sync_sema = xSemaphoreCreateBinary();
        assert(at_sync_sema != NULL);
    }

    // 返回输入数据提示符 ">"
    esp_at_port_write_data((uint8_t *)">", strlen(">"));

    // 设置回调函数，在接收到输入数据后由 AT 端口调用
    esp_at_port_enter_specific(wait_data_callback);

    // 接收输入的数据
    while(xSemaphoreTake(at_sync_sema, portMAX_DELAY)) {
        received_len += esp_at_port_read_data(buf + received_len, specified_len - received_len);

        if (specified_len == received_len) {
            esp_at_port_exit_specific();

            // 获取剩余输入数据的长度
            remain_len = esp_at_port_get_data_length();
            if (remain_len > 0) {
                // 示例代码
                // 如果剩余数据长度 > 0，则实际输入数据长度大于指定的接收数据长度
                // 可自定义如何处理这些剩余数据
                // 此处只是简单打印出剩余数据
                esp_at_port_recv_data_notify(remain_len, portMAX_DELAY);
            }

            // 示例代码
            // 输出接收到的数据
            memset(buffer, 0, 64);
            snprintf((char *)buffer, 64, "\r\nreceived data is: ");
            esp_at_port_write_data(buffer, strlen((char *)buffer));

            esp_at_port_write_data(buf, specified_len);

            break;
        }
    }

    free(buf);

    return ESP_AT_RESULT_CODE_OK;
}

ustccw, 2023-06-05:
@xiaoheng14 没太懂你描述的问题和代码之间的关联。可否详细说下。是在哪一行代码上，出现和预期不符合的结果呢。以及对应的操作。

xiaoheng14, 2023-06-05:
好的，是在接收输入数据的时候，`received_len += esp_at_port_read_data(buf + received_len, specified_len - received_len);`, 这里应该是一直循环等待接收数据直到读取到specified_len长度的数据的，但是我在测试的时候发现要是数据分两次发送，总是会在第一次就完成了，导致buf的数据后面的部分是未填充的。比如发送指令`AT+TEST=10`，然后`>`返回后先发送部分数据，`hello`，然后直接就返回`SEND OK`了，读取到的数据变成了`hello\0\0\0\0\0`或者乱码。但是使用`AT+CIPSEND`这个指令就不会这样。

ustccw, 2023-06-05:
@xiaoheng14 我用你的代码测试了下，不会呀
[image]
。

测试代码在 https://github.com/ustccw/esp-at/commit/a6f77d45a2bd602e87cd2d4ecf4b1c6f75250555
对应日志只在 673 行添加了一个 AT port 口的打印：
```
printf("received_len:%d (expect len:%d)\r\n", received_len, specified_len);
```

你再检查下代码呢，再抓下 uart 线上数据看看。

xiaoheng14, 2023-06-05:
好的，我再检查看看，麻烦您了

ustccw, 2023-06-05:
好的，不客气~

closed 2023-07-17 by ustccw