espressif/esp-at #725: BLE services and characteristics fail to initialize

JakaKor, 2023-05-22:
Hello all,

I have some problems with creating BLE services. I tried with the ESP32-WROOM and ESP32C3-MINI. 
First I tried with the latest version. I compiled and flash the project successfully, but I couldn't get the BLE services to start. 

```
AT+GMR
AT version:3.1.0.0-dev(s-bd890df - ESP32 - May 12 2023 08:47:33)
SDK version:v5.0-541-g885e501d99
compile time(c5af3bc):May 15 2023 12:00:57
Bin version:2.5.0(WROOM-32)

OK
```

When I try to set up the BLE I get the following:

```
AT+BLEINIT=2

busy p...

OK

AT+BLEGATTSSRVCRE

busy p...

OK

AT+BLEGATTSSRV?

busy p...
+BLEGATTSSRV:1,0,0x,1
+BLEGATTSSRV:2,0,0x,1

OK

```

I can connect to ESP with my phone, but there isn't any service or characteristics visible.
I think I get similar problem as described in issue #696

My solution was that I build the 2.4 release for ESP AT (5641e0aa2b3cbc36b2d843a72dd00d248daaef40), but I was unable to compile it with the ESP-IDF 5.0. 
Probably because of the different versions?

```
branch:release/v4.3
commit:588436db47da792736dade1c21fcbf2c879456a4
repository:https://github.com/espressif/esp-idf.git
```

```
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.0_py3.11_env
    Python interpreter used: C:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe

ESP-IDF v4.3.2-553-g588436db47
A fatal error occurred: idf.py build failed
```

I tried to compile it with ESP-IDF 4.3, without changing the branch I get similar error:

```
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf4.3_py3.11_env
    Python interpreter used: C:\Espressif\python_env\idf4.3_py3.11_env\Scripts\python.exe

ESP-IDF v4.3.2-553-g588436db47
A fatal error occurred: idf.py build failed
```

Later I changed the commit to (33b3db35d73d60d12dae52157b2f5c5dbe1a813c)

build and flash are successful.

serial monitor snippet:

```
AT+GMR
AT version:2.4.1.0(s-fb731d2 - ESP32 - Aug 25 2022 03:48:17)
SDK version:v4.3.5-83-g33b3db35d7
compile time(f64aa35):May 15 2023 13:19:23
Bin version:2.4.1(WROOM-32)

OK
AT+BLEGATTSSRVCRE

busy p...

OK
AT+BLEGATTSSRV?

busy p...
+BLEGATTSSRV:1,0,0xA002,1
+BLEGATTSSRV:2,0,0xA003,1

OK
AT+BLEGATTSSRVSTART=1

busy p...

OK
AT+BLESECPARAM=1,3,16,3,3,0

busy p...

OK
AT+DRVI2CINIT=0,19,18,100000

ERROR
```

Thank you for your help.

ustccw, 2023-05-22:
@xiewenxiang could you please help with this issue.

JakaKor, 2023-05-28:
Is there any solution to this problem? Or there is something wrong with my approach?

xiewenxiang, 2023-05-29:
@JakaKor  We will fix it as soon as possible.

3s1d, 2023-06-11:
+1

Any updates on that topic? I am facing the same issue.  See #728 (which I closed already). Please fix it.

Luxin812, 2024-03-20:
@JakaKor @3s1d 
This issue has been fixed and incorporated into the master. You can update your AT firmware to the latest. Test to see if the issue has been fixed. If it has been fixed, please close the issue. Thank you. (https://github.com/espressif/esp-at/actions/runs/8294336044)

closed 2024-07-29 by ustccw