espressif/esp-at #727: atPKI is not available in main repository

MaJerle, 2023-06-10:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Why not? While it is advertised here: https://docs.espressif.com/projects/esp-at/en/release-v2.4.0.0/esp32/Compile_and_Develop/how_to_generate_pki_files.html#generate-certificate-bin-files

https://github.com/espressif/esp-at/blob/f64aa352/tools/AtPKI.py

ustccw, 2023-06-12:
@MaJerle Hello~
The `ATPKI.py` is a deprecated script in master branch, and it's still available in release/v2.4.0.0 branch: [ATPTK.py](https://github.com/espressif/esp-at/blob/release/v2.4.0.0/tools/AtPKI.py). 
In short, the recommended version of the document is consistent with the version of the code repository. If you use [release/v2.4.0.0 branch](https://github.com/espressif/esp-at/tree/release/v2.4.0.0.0), then you should read the [release-v2.4.0.0 document](https://docs.espressif.com/projects/esp-at/en/release-v2.4.0.0/esp32/index.html).

### 1. Why was ATPKI.py deleted on the master branch?
Because the parameter storage mode changed from [old partition](https://github.com/espressif/esp-at/blob/release/v2.4.0.0/module_config/module_esp32_default/at_customize.csv) way (old version) to [manufacturing nvs](https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/at_customize.csv) way (new version). The `ATPKI.py` is only applicable for the old partition storage mode. So we removed it on master branch.

### 2. Why are we using  [manufacturing nvs](https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/at_customize.csv) way on the master?
- **Low usage rate of flash**
   The partition occupies is large, resulting in low effective utilization of this portion of the flash.
   For example 1: The certificate partitions are planned with a size of 8KB or 4KB each, but in reality, only 1KB+ is needed.
   For example 2: Some partitions are not used by customers but still occupy a significant amount of flash, such as certificates for connecting to enterprise-ap routers.
- **Lack flexibility**
   For example 1: For mass-produced products, different modules may have different configurations (different certificates, private keys, etc.), making it impossible to directly generate differentiated firmware binaries for different modules.
   For example 2: If a customer wants to expand the configuration, they must reorganize the secondary partition table and recompile.
- **Risk of over-erasure**
   For example: Due to non-robust code, frequent updates to a specific configuration result in erasing and writing a sector every time through partition read/write at a fixed address, which reduces the lifespan of the flash. However, if the new nvs way is used for read/write operations, the number of sector erasures can be significantly reduced, thereby extending the lifespan of the flash.

Based on the above considerations, starting from v3.1.0.0 and later versions, the [manufacturing nvs](https://docs.espressif.com/projects/esp-at/en/latest/esp32/index_of_abbreviations.html#term-manufacturing-nvs) way will be used instead of the partition way to store configurations in the secondary partitions. This allows for dynamic flash allocation and facilitates scalability requirements.

### 3. Is it compatible with older versions?
Yes, in almost cases. The firmware will prioritize reading the ``mfg_nvs`` partition. If the partition is found, the firmware will read the parameters using the new `manufacturing nvs` storage method. If the partition is not found, the firmware will fall back to the old `partition` storage method for reading.
There is just a known breaking change, described here: https://github.com/espressif/esp-at/releases/tag/v3.1.0.0#breaking-changes

Please know, thanks.
~

MaJerle, 2023-06-13:
When we can expect 3.1 for all ESP32xx modukes?

ustccw, 2023-06-13:
> When we can expect 3.1 for all ESP32xx modukes?

Yes, we are preparing version v3.2.0.0 for ESP32 and ESP32-C3. It is currently in testing phase and expected to be released in early July.

MaJerle, 2023-07-28:
I've seen the release v3.2 for ESP32. How shall we now setup the custom certificate file to be loaded to the device? Any tutorial on that available?

ustccw, 2023-07-31:
Hello @MaJerle,

**1).** In the runtime, MCU sends [AT+SYSMFG](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#at-sysmfg-query-set-manufacturing-nvs-user-partitions) command to update certificates or keys, or any other configurable parameters. In terms of functionality, a crude understanding would be like: `AT+SYSMFG` ~= (`AT+SYSFLASH` + `ATPKI.py`) 

This way requires MCU to send AT command, beside it, users could consider more ways to update this kind of parameters:

**2).** Modify the released firmware by using [at.py tool](https://github.com/espressif/esp-at/blob/master/tools/at.py), refer to [Detailed Steps](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/tools_at_py.html#at-py-tool) for more details.

**3).** In the compile stage, users can refer to [How to Update PKI Configuration](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_update_pki_config.html) to update these configurations.

MaJerle, 2023-07-31:
3. What about at runtime? So that we can keep offi ual AT firmware release as is?

ustccw, 2023-07-31:
sorry that I didn't catch up your point, could you please describe it in details?

closed 2023-10-03 by MaJerle