espressif/esp-at #729: Connection closed before all data read.

MaJerle, 2023-06-22:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

2 issues found during development of the application. Device is in manual receive mode.

## Issue 1: +IPD returns info, but connection gets closed and read is not possible.

Why don't we wait and let user to first read data before closed is reported? Or let him read after it is closed?

```
+IPD,0,31
AT+CIPRECVLEN?
+CIPRECVLEN:31,-1,-1,-1,-1

OK
AT+CIPRECVLEN?
+CIPRECVLEN:31,-1,-1,-1,-1

OK
AT+CIPRECVDATA=0,31

ERROR
0,CLOSED
AT+CIPRECVLEN?
+CIPRECVLEN:-1,-1,-1,-1,-1

OK
```

## Issue 2: +IPD and actual read mismatch.

Why there is no missing `29` bytes?

```
+IPD,0,523
AT+CIPRECVLEN?
+CIPRECVLEN:523,-1,-1,-1,-1

OK
AT+CIPRECVDATA=0,523
+CIPRECVDATA:494,"13.107.246.45",443,....actual data....

OK
AT+CIPRECVLEN?
+CIPRECVLEN:0,-1,-1,-1,-1

OK
```

ustccw, 2023-06-25:
Hello @MaJerle,
If this is the TCP connection, it must be two bugs. If this is the SSL connection, it should be two reasonable cases.

Let me try to clarify this two issues if use SSL connection.
### Issue 1: +IPD returns info, but connection gets closed and read is not possible.
could you please check the log port print.
if it prints `NO DATA`, it means no ssl data. it might receive a ssl alert packet from the server. and `AT+CIPRECVLEN?` returned 31 bytes, it indicates that tcp data length is 31 bytes, not the ssl data. this tcp data may be that alert packet.
Otherwise, there is no more memory for `calloc` and output the data.

### Issue 2: +IPD and actual read mismatch.
Similar to https://github.com/espressif/esp-at/issues/589, https://github.com/espressif/esp-at/issues/655

In short, for SSL connections, ESP-AT will prioritize returning the data length of the ssl layer. if the ssl layer length is 0, it will return the data length which buffered at the tcp layer (Typically, this means that the ssl layer has not yet received a complete ssl fragment and is waiting to receive new TCP data).

MaJerle, 2023-06-25:
It was for SSL connection, yes.

Thanks for clarification.

ustccw, 2023-06-26:
no worries

closed 2024-10-10 by ustccw