espressif/esp-at #730: esp-at: reduce firmware size

amiclaus, 2023-07-04:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to enable AT command support both BT and WiFi simultaneously. The issue is that the flash size that I have at disposal is 1MB. Is there a way to reduce the firmware size below 1MB in this particular situation? I enabled all optimizations -Os and removed all unnecessary AT features:
[image]

 But still cannot go below 1.1MB for the generated binary. (esp-at.bin). 

Any help is appreciated.

ustccw, 2023-07-11:
@amiclaus Helllo, you can reduce the firmware size according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html#minimizing-binary-size guide.
For esp-at project, you can enable silent mode at step 3 when you compile the esp-at project.

~

amiclaus, 2023-07-14:
Thanks for the feedback!

closed 2023-07-20 by amiclaus