espressif/esp-at #731: busy p... received after previous command returns OK

MaJerle, 2023-07-08:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I can often see a `busy p...` response to a command, even if previous one returned OK. Why is that?

```
AT+CIPSEND=0,4268

OK

>
Recv 4268 bytes

SEND OK

+IPD,0,654
AT+CIPRECVLEN?
+CIPRECVLEN:654,-1,-1,-1,-1

OK
AT+CIPRECVDATA=0,654
ERR CODE:0x010b0000   <------------- Indicates sub command is executing?

busy p...             <------------- WHY BUSY P HERE?
+CIPRECVDATA:625,"13.69.186.152",443,HTTP/1.1 200 OK
Content-Length: 0
Connection: close
...the rest of the CIPRECVDATA

OK
0,CLOSED
```

MaJerle, 2023-07-08:
Do I need a delay between any specific sequence somewhere? 

It is the latest firmware for ESP32.

ustccw, 2023-07-10:
@MaJerle From the description you provided, if the commands are dead right and no more extra chars appended to the commands, it should be a bug. could you please help to reproduce this issue with the following debug libraries.

[libespat.zip](https://github.com/espressif/esp-at/files/11997589/libespat.zip)

many thanks.

@Luxin812 please help to add testcases to reproduce this issue, thanks.

MaJerle, 2023-07-10:
Any chance you build that for me since I have no tools? For WROOM-32 and WROVER-32, please. WIth factory AT.

Another point, if you looks at example 2, why is there response with data I wanted to send, just after "SEND OK" has been dispatched? Potential buffer overflow somewhere?

ustccw, 2023-07-10:
@MaJerle the firmware is in the attachment -> build -> factory -> factory_xxx.bin, please have a try, and provide the uart0 logs (logs from download port), thanks.

[ghi-731.zip](https://github.com/espressif/esp-at/files/11998526/ghi-731.zip)

about example 2, I checked the code. and doubt there is more data before `>` prompt. I'm always confused this.
if this is can be easy to reproduced, I would like to fix it asap.

MaJerle, 2023-07-10:
I am able to reproduce this, but system has to run for a while -> We will load your firmware and report AT and uart0 logs.

MaJerle, 2023-07-10:
Testing firmware now. I see it is v3.2.0, which is newer than used before. Do you have it with the latest 2.x version for ESP32 ?

MaJerle, 2023-07-10:
Here we go. 2 files are attached, I have removed the company confidential information.
I have been using *Termite* UART terminal, and have enabled the Timestamp feature. That's the reason you see sometimes time in between the reply, such as:

```
AT+CIPSEND=0,8192

OK

>
Recv 8192 bytes
2023/07/10 20:22:25: 
SEND OK
```
Same timestamp is available for both files, it can be used to sync the debug.

[firmware_at_commands_response.txt](https://github.com/espressif/esp-at/files/12005168/firmware_at_commands_response.txt)
[firmware_debug_port_response.txt](https://github.com/espressif/esp-at/files/12005169/firmware_debug_port_response.txt)

I will continue to run the example for more cases.

ustccw, 2023-07-11:
> Testing firmware now. I see it is v3.2.0, which is newer than used before. Do you have it with the latest 2.x version for ESP32 ?

lastest https://github.com/espressif/esp-at/tree/release/v2.2.0.0_esp32 branch does not backport many bugfixes, including `busy p` issue. it is recommended to use lastest master branch (or release/v3.1.0.0 branch) for esp32 chip now.

ustccw, 2023-07-11:
> Here we go. 2 files are attached, I have removed the company confidential information.
> I have been using _Termite_ UART terminal, and have enabled the Timestamp feature. That's the reason you see sometimes time in between the reply, such as:

It seems that no `busy p` printout was seen. Could it be that the issue with `busy p` did not reproduce?

MaJerle, 2023-07-11:
You are right, in this case I was able to reproduce only the issue when data to send are returned on the AT port, which should not be the case I guess.

> lastest https://github.com/espressif/esp-at/tree/release/v2.2.0.0_esp32 branch does not backport many bugfixes, including busy p issue. it is recommended to use lastest master branch (or release/v3.1.0.0 branch) for esp32 chip now.

We only rely on the official release of AT commands by Espressif. We do not make builds by ourselves. Therefore waiting for v3.2 for ESP32.

ustccw, 2023-07-31:
Hello @MaJerle , ESP-AT released v3.2.0.0 firmwares for ESP32 and ESP32-C3 last week, please know.
https://github.com/espressif/esp-at/releases/tag/v3.2.0.0

ustccw, 2024-03-07:
We assume this worked for you, and will close it. If that's not the case, please open another github issue.

mzirkle44060, 2024-06-20:
What was the final solution on this? I'm building and deploying esp-at (v.4.0.0.0) for the C6 and I'm constantly receiving the busy p... in every response?
[AT]    CMD: ATE0
[AT]    RSP: ATE0

busy p...

OK

[AT]    CMD: AT+UART_CUR=115200,8,1,0,1
[AT]    RSP:
busy p...

OK

ustccw, 2024-07-01:
It should not happen again, could you please provide more information according to [this template](https://github.com/espressif/esp-at/issues/new?assignees=&labels=Type%3A+Bug&projects=&template=02_runtime_bug.yml).