espressif/esp-at #732: esp32c3 怎么增加使用spi-at时httpclient指令每次握手传输数据大小

ctzxc88, 2023-07-09:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

主机使用stm32 spi 接近37MHZ 从机使用esp32c3，使用spi-at方式，但httpclient指令从机每次最多传输512字节，在传输数据时就出现了需要频繁握手的问题，请问我该在哪增加这个httpclient指令的单次握手传输字节的大小？
另外，我在使用官方的esp-at-2.4.2.0.zip固件尝试set-target为esp32c3时报错：
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at esp-idf/tools/cmake/project.cmake:416 (message):
  SDKCONFIG_DEFAULTS
  'C:/Espressif/frameworks/esp-idf-v5.0.1/esp-at/module_config/module__default/sdkconfig.defaults'
  does not exist.
Call Stack (most recent call first):
  CMakeLists.txt:91 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v5.0.1/esp-at/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the c:\espressif\frameworks\esp-idf-v5.0.1\esp-at\build\log\idf_py_stderr_output_10612 and c:\espressif\frameworks\esp-idf-v5.0.1\esp-at\build\log\idf_py_stdout_output_10612

C:\Espressif\frameworks\esp-idf-v5.0.1\esp-at>
可以请问下是怎么回事吗？

ustccw, 2023-07-10:
1. 请考虑使用 AT+HTTPCGET 命令：https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32c3/AT_Command_Set/HTTP_AT_Commands.html#cmd-httpgetsize。 
否则你需要自己编译，修改源码：
`esp-idf/components/esp_http_client/include/esp_http_client.h` 中的 DEFAULT_HTTP_BUF_SIZE 大小。

2. 编译请参考：https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32c3/Compile_and_Develop/How_to_clone_project_and_compile_it.html

ctzxc88, 2023-07-18:
感谢您的回答！
我找到了这个宏。关于这个宏我想问您几个问题：
①在用户指南的spi at指南中有个TCP透传 8192字节，这个透传大小和 httpclient指令获得的数据大小有关系吗？
②只用更改这个宏，其他地方不用更改，程序就会在存满这个buffer的时候向MCU发送数据吗?
③在driver的spi中有最大4092的限制，我是否可以将这个宏和spi最大4092都改为32767或更大的数字?

ustccw, 2023-07-28:
@ctzxc88 
1. 不直接相关，看下 [如何提高 ESP-AT 吞吐性能](https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32c3/Compile_and_Develop/How_to_optimize_throughput.html#id2) 文档，SPI-AT 的 buffer 在 S1/R1 阶段，HTTP client 的 buffer 大致在 G9 阶段（准确说在 HTTP 层的 buffer）。
2. 不会存满才会向 MCU 发送。这是最大 buffer，一旦收到数据，都会向 MCU 发送。
3. 不能，也没必要。具体参考 [SPI Slave Mode Protocol](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/protocols/esp_spi_slave_protocol.html)。
4. 在环境安装好后，ESP-IDF 可以离线使用。

closed 2023-10-30 by ctzxc88