espressif/esp-at #751: CWAUTOCONN, where are AP list stored and how to remove it?

MaJerle, 2023-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm looking at various places in the docs, but don't seem to find:

- `AT+CWAUTOCONN` only talks about auto connection to access point
- `AT+CWJAP` does not mention anything about storing AP info to internal memory
- `AT+CWQAP` also doesn't mention anything about storage of automatic connection

Questions:
- How can we list "preferred" access points from the ESP, that are used as auto connection? At least SSID or MAC is needed
- How can we delete preferred access point entries (one by one is OK, so that we can filter)

Thanks

JAndrassy, 2023-08-12:
sorry, do you want to know these things or you know them but you ask Espressif to add it to docs?
because I have some partial answers if you are interested before there is an official response from Espressif employees

MaJerle, 2023-08-12:
Nono, I do not know the answers. If you do, please elaborate :)

JAndrassy, 2023-08-13:
As I understand it from docs and experience, only one AP is stored.

the AT 1 reference pdf had a section about it. it described that WiFi configuration is stored by SDK and other parameters are stored by the AT firmware itself.

In esp32 AT docs I found this:
https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/index.html#at-commands-with-configuration-saved-in-the-flash
and from there this
https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-sysstore

from AT 1 I know that the AP data are stored by the framework, it is IDF for AT 2 so U searched the docs for IDF:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-nvs-flash

Maybe it is possible to use AT+SYSMFG to view and delete the WiFi configuration. I didn't try it.
https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html?highlight=nvs#cmd-sysmfg

ustccw, 2023-09-01:
@MaJerle @JAndrassy Hello, here are some comments about this issue.
1. ESP-AT just stores one AP information now, which by [esp_wifi_set_storage()](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h#L876-L887) API. This configuration (or similar configuration) is stored at [nvs partition](https://github.com/espressif/esp-at/blob/master/module_config/module_esp32c3_default/partitions_at.csv#L4). as @JAndrassy mentioned, it's controlled by [AT+SYSSTORE](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-sysstore) command.
2.  You can use [AT+CWSTATE?](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Wi-Fi_AT_Commands.html#cmd-wstate) command to query the ssid and state. (no bssid)
3. AT cannot delete preferred access point entries for nvs partition directly, unless a new configuration is set through `AT+CWJAP` or a new command is added to implement it.
4. `AT+SYSMFG` cannot delete the Wi-Fi configurations. because `AT+SYSMFG` commands operate the [mfg_nvs](https://github.com/espressif/esp-at/blob/master/module_config/module_esp32c3_default/at_customize.csv#L2) partition (see [mfg_nvs docs](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_update_mfg_nvs.html) for more details), while Wi-Fi configuration is stored in  [nvs partition](https://github.com/espressif/esp-at/blob/master/module_config/module_esp32c3_default/partitions_at.csv#L4).

we should add more docs to clarify this, thank you!

MaJerle, 2023-09-01:
Is there a way to at least delete entries with AT commands, or can we add some?

ustccw, 2023-09-04:
If i'm not mistaken, what you want is to delete these configs from nvs. May i know what's the last goals for the real products. If you don't want to save the configs, you can send `AT+SYSSTORE=0` command before `AT+CWJAP`, or control the Wi-Fi connection by `AT+CWQAP`, `AT+CWJAP`, `AT+CWRECONNCFG`, `AT+CWAUTOCONN` commands.

Besides, if it's necessary for you, there is a feasible recipe to add an AT command to erase the key-value pairs from nvs partition. To know the `namespace` name and `key` name, you can enable the debug logs on menuconfig:
`python build.py menuconfig` -> `Component config` -> `Log output` -> `Default log verbosity` -> `Debug`, and then the logs should be like the following. You can erase the ssid or similar configurations by `nvs_erase_key("sta.ssid");`, etc.

```
D (1509) nvs: nvs_open_from_partition nvs.net80211 1
D (1514) nvs: nvs_get opmode 1
D (1518) nvs: nvs_get_str_or_blob sta.ssid
D (1522) nvs: nvs_get sta.authmode 1
D (1525) nvs: nvs_get_str_or_blob sta.pswd
D (1530) nvs: nvs_get_str_or_blob sta.pmk
D (1533) nvs: nvs_get sta.chan 1
D (1537) nvs: nvs_get auto.conn 1
D (1540) nvs: nvs_get bssid.set 1
D (1543) nvs: nvs_get_str_or_blob sta.bssid
D (1548) nvs: nvs_get sta.lis_intval 2
D (1552) nvs: nvs_get sta.phym 1
D (1555) nvs: nvs_get sta.phybw 1
D (1558) nvs: nvs_get_str_or_blob sta.apsw
D (1563) nvs: nvs_get_str_or_blob sta.apinfo
D (1568) nvs: nvs_get sta.scan_method 1
D (1571) nvs: nvs_get sta.sort_method 1
D (1575) nvs: nvs_get sta.minrssi 1
D (1578) nvs: nvs_get sta.minauth 1
D (1582) nvs: nvs_get sta.pmf_e 1
D (1585) nvs: nvs_get sta.pmf_r 1
D (1588) nvs: nvs_get sta.btm_e 1
D (1592) nvs: nvs_get sta.rrm_e 1
D (1595) nvs: nvs_get sta.mbo_e 1
```

~

closed 2024-03-18 by ustccw