espressif/esp-at #755: ESP8266EX_Demo_Boardno exchange of AT commands via USB

edbek, 2023-08-12:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

no

### ESP-AT Firmware Source

no

### Hardware Information

ESP8266EX_Demo_Board (a chip ESP8285 is installed on the board).

### Power Supply used

USB

### What is the expected behavior?

After flashing the finished image from "esp8285-1MB-at.zip" (factory_ESP8266_1MB.bin) into the device ESP8285, I do not go out to exchange AT commands over the same USB channel channel through which I flashed the device.
Does anyone know why this is happening?

### What is the actual behavior?

no AT command exchange.

### Probability of recurrence

always

### AT+SYSRAM?

no

### Steps to reproduce

1. Step
2. Step
3. Step
...
no

### AT command port output

```plain
absent
```

### AT log port output

```plain
absent
```

### More Information.

no

JAndrassy, 2023-08-12:
the AT 2 uses UART on pins 15 as TX and 13 as RX for AT commands. To have the AT firmware communicating on usual pins 1 as TX and 3 as RX, you can use  instead of factory_param.bin bundled with the AT firmware the [_esp8266_factory_param_tx1rx3.bin](https://github.com/JAndrassy/WiFiEspAT/tree/01a72cb2335f5db69588c80bc22b7c0b532ac3e5/extras) provided in the past as extra with my library.

btw: the AT 1.7.5 from NonOS SDK 3.0.5 is much better for ESP8266/ESP8285

edbek, 2023-08-12:
Thank you !

But how confusing. The company itself suggests using RTOS SDK instead of the NonOS SDK. 
And I have a question about command sets: does the NonOS SDK support the following commands that I need ?:

AT+CWLAPOPT: Set the Configuration for the Command AT+CWLAP
AT+CWLAP: List Available APs

I can't find the AT Instruction Set for the NonOS SDK.

edbek, 2023-08-12:
AT version to 1.7.5 for NonOS SDK is very old (Oct 15, 2021).
It's probably better to use the new AT version for the RTOS SDK.

JAndrassy, 2023-08-13:
yes AT 1 has both commands. and AT 1 is very stable and robust.
there is no newer version of AT 2 for esp8266.
AT 2 for esp8266 is discontinued and the available version 2.2 of AT 2 for esp8266 are unreliable

edbek, 2023-08-14:
I flash the AT 1.7.5 from NonOS SDK 3.0.5 device ESP8285, but I see the same thing: I do not go out to exchange AT commands over the same USB channel.

ustccw, 2023-09-01:
oh, no. i assume that you have already read the [support policy](https://github.com/espressif/esp-at#esp-at-support-policy-for-esp-chip-series) about ESP8266/ESP8285 for AT 2 and [support policy](https://github.com/espressif/ESP8266_NONOS_SDK#support-policy-for-esp8266-nonos) for AT 1.

If you are ready to start a new project, please consider ESP32-C2 or ESP32-C3 chip instead.

AllenCyborg, 2023-09-26:
had the same issue and used the esp01s binary from: https://github.com/CytronTechnologies/esp-at-binaries on my esp01. UART is working

> yes AT 1 has both commands. and AT 1 is very stable and robust. there is no newer version of AT 2 for esp8266. AT 2 for esp8266 is discontinued and the available 2021 version 2.2 of AT 2 for esp8266 is unreliable
> 
> the documentation is "ESP-AT Instruction User Guide" at https://www.espressif.com/en/support/documents/technical-documents?keys=&field_type_tid%5B%5D=14&field_download_document_type_tid%5B%5D=965

@JAndrassy how is v2.2.x unreliable?

JAndrassy, 2023-09-26:
> @JAndrassy how is v2.2.x unreliable?
these are the issues I could describe:
https://github.com/espressif/esp-at/issues/458
https://github.com/espressif/esp-at/issues/655

closed 2024-03-18 by ustccw