espressif/esp-at #759: Large file upload

VictorDinh2105, 2023-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How do I upload a video file which is too big to attach to email and send out for support.

VictorDinh2105, 2023-09-14:
When I upload here, it is limited to 10MB.  My file size is 33 MB.

ustccw, 2023-10-07:
try to put that video on GitHub repository or any http/https server, and paste the url here.

closed 2024-03-18 by ustccw