espressif/esp-at #763: Easy_install not found MACOS Ventura

LLC09, 2023-10-03:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html) and the issue is not addressed there.
- [X] I have compiled the hello_world example [Get Started with ESP-IDF](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#step-1-get-started-with-esp-idf) and the issue is not addressed there.
- [X] I have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### ESP-AT version

V3.2.0.0

### Operating System used

macOS

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

According to the guide, to perform the installation it needed to run the build.py script as follows into the terminal:

./build.py install

And the output will be all the stuff information regarding the installations.

### What is the actual behavior?

Instead, it returns an error as follows

install prerequisites failed! Please manually run: sudo easy_install pip && brew install cmake ninja dfu-util ccache python3

### Steps to reproduce

1. Launch ./build.py on macos Ventura

### Build or installation Logs

```plain
I have not
```

### More Information

To fix the issue described I commented on the check to install the prerequisite in the main function(lines 520 and 521). It made the work and the script performed the installation (Installing modules, step by step. According to missing package error output). Of course, it's not a clean fix, to perform that I suggest checking if easy_install is present, otherwise perform the installation through brew.
Another thing,
sudo easy_install pip && brew install cmake ninja dfu-util ccache python3
as you can see here is called python3 but It's wrong or at least, aligned with the other scripts, it should be called 'python' instead of 'python3' .
I suggest installing pyenv and handle with different python versions easily.

ustccw, 2023-10-07:
@LLC09 you are right, could you replace line 474, and have a try.
```
cmd = 'sudo easy_install pip && brew install cmake ninja dfu-util ccache python3'
```
with
```
cmd = 'brew install cmake ninja dfu-util'
```

LLC09, 2023-10-07:
@ustccw It's a good idea, but it doesn't work due to lack of permission this is the output.
[image]
After giving the required permissions I got this one
[image]
I think it's possible to try to give the 'sudo' when the script calls the script "__about__.py", in general, I think it's possible to ask for permission each time it is needed.
Unfortunately isn't possible to give permission to the script build.py due to brew not being executed with root privileges.

ustccw, 2023-10-08:
what if run the following commands:
```
cd esp-at/esp-idf

./install.sh

. ./export.sh

cd ..

python -m pip install -r requirements.txt

python build.py build
```