espressif/esp-at #767: AT BluFi missing Bluetooth connection and disconnect URC message

xiaoxinpro, 2023-10-09:
### Is your feature request related to a problem?

The serial port information of AT BluFi is as follows:

```
AT+BLUFI=1

AT+BLUFI=1
OK

+BLUFIDATA:6,123456
```

The following contains Log information:

```
[发送][14:56:10.060]AT+BLUFI=1

[接收][14:56:10.160]AT+BLUFI=1

[Log][14:56:10.251]BLUFI init finish

[接收][14:56:10.257]
OK

[Log][14:57:04.999]BLUFI ble connect

[接收][14:57:34.481]+BLUFIDATA:6,123456

[Log][14:57:37.845]blufi close a gatt connectionBLUFI ble disconnect

[Log][14:58:19.529]BLUFI ble connect

[Log][14:58:32.020]BLUFI get wifi status from AP

```

### Describe the solution you'd like.

It is expected that the corresponding URC message can be sent when the Bluetooth is connected and disconnected

for example:

```
AT+BLUFI=1

AT+BLUFI=1
OK

+BLECONN

+BLUFIDATA:6,123456

+BLEDISCONN
```

### Describe alternatives you've considered.

No alternative has been found

### Additional context.

This function is used to check whether the APP Bluetooth is connected to the ESP32 module

ustccw, 2024-01-22:
@Luxin812 please check

Luxin812, 2024-03-20:
@xiaoxinpro The latest master branch already supports printing blufi connection and disconnection logs. You can update the firmware to the latest and test it.

xiaoxinpro, 2024-03-20:
@Luxin812 Pass test, Thanks.

Here is the test log:

```
[发送] AT+BLUFI=1

[接收] AT+BLUFI=1

[接收] +BLUFICONN

[接收] +BLUFIDATA:6,123456

[接收] +BLUFIDISCONN
```

closed 2024-03-21 by ustccw