espressif/esp-at #768: ESP32 AWS Fleet Provisioning with AT Commands Problem

yukselen26, 2023-10-09:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

We are trying to perform AWS fleet provisioning on an ESP32 C3-Mini using AT commands. When we place the claim certificate in the appropriate directories (esp-at\components\customized_partitions\raw_data) and flash it with ESP-IDF, we can establish a connection without any problems. However, when we embed the same certificate into the ESP32 using the AT+SYSMFG command and then attempt MQTT connection, it fails. Even though we see the exact same string when we embed and read the certificates using AT commands, we still cannot establish a connection. It seems there might be an issue with data format or something else, but we can't figure out what it is. Are we inadvertently altering any structures when we copy and send the certificate text via the terminal?

Any assistance would be greatly appreciated.

gavin-hy, 2023-10-17:
I also encountered the same problem.Certificate and key sent using AT+SYSMSG command.
[image]
[image]

yukselen26, 2023-10-18:
Problem solved. The problem was solved when I selected binary instead of string when installing the certificate (AT+SYSMFG)

yukselen26, 2023-10-18:
> I also encountered the same problem.Certificate and key sent using AT+SYSMSG command. [image] [image]

Try flashing the certificate with AT+SYSMFG. Also pay attention to the format of the data, you must move to the next line every 64 characters (\n). When you read the data, \n should not be visible, but you should see that it moves to the next line every 64 characters.

closed 2024-07-29 by ustccw