espressif/esp-at #772: FR: HTTP interact with the file system

pauldeng, 2023-10-20:
### Describe the solution you'd like.

It would be great if HTTP commands could work with the file system.

For example:
 - AT+HTTPCGET could straight download the file into ESP32 file system
 - AT+HTTPCPOST and AT+HTTPCPUT could straight post/put the specified file from ESP32 file system to a url

This feature is similar to the [AT+UHTTPC command](https://cdn.sparkfun.com/assets/b/6/2/4/5/SARA-R5_ATCommands__UBX-19047455_.pdf) provided by the U-Blox R510S module.

pauldeng, 2023-10-31:
Dear repo owner/maintainer,

Is there any update for this Feature Request?
We probably could have a discussion about it.

Cheers,
Paul

ustccw, 2023-11-03:
@pauldeng thanks for your feature request. We will consider it with low priority because the default flash available is very limited and can only store very small files.

pauldeng, 2023-11-04:
Dear @ustccw, 

Thanks for your reply.

This feature is not intended to be used to download a movie. In my use case, it is a binary file that is < 1MB.

Can I ask how to prioritise this feature? Or do you allow contribution from other personal/company, so that I probably can work on this.

We can discuss how to proceed if you are interested.

Cheers,
Paul

pauldeng, 2023-11-20:
Dear @ustccw,

If I would like to contribute, how do I proceed?

Cheers,
Paul

ustccw, 2024-01-15:
@pauldeng I apologize for the delayed response. If you urgently need this feature, it might be more appropriate for you to undertake secondary development and implement custom AT commands. We also have plans for this feature, but it's scheduled for 2024.Q2.

pauldeng, 2024-01-18:
That is great!
Appreciate.

closed 2024-01-18 by pauldeng