espressif/esp-at #773: FR: File integrity check command

pauldeng, 2023-10-20:
### Describe the solution you'd like.

It would be great if a new AT command is provided to compute the checksum of a file in the file system.

For example:
`AT+FSCRC=<filename>,<algorithm>`
This command will compute the checksum of the specified file using the specified algorithm such as crc-16-modbus, md5, sha256...

When done, it prints out the hash code.
```
+FSCRC:<filename>,<algorithm>,<hashcode>
OK
```

This will be useful to check the integrity of a file.

pauldeng, 2023-10-31:
Dear repo owner/maintainer,

Is there any update for this Feature Request?
We probably could have a discussion about it.

Cheers,
Paul

ustccw, 2023-11-03:
@pauldeng Hello, may I ask which file do you want to perform integrity verification on. the file from an "URL" or from MCU side, or file from AT+FS command?

pauldeng, 2023-11-04:
Dear @ustccw,

Thanks for your reply.

I would like to verify the files only locally within ESP32 file system (NOT a URL or from MCU).
Basically, any file that can be operated by `AT+FS` command.

The verification algorithm can be any ESP32 hardware supported.

In my use case, I need to download a binary OTA file from URL to ESP32 file system, and verify its integrity before the MCU apply it.

Cheers,
Paul

pauldeng, 2023-11-20:
Dear @ustccw,

If I would like to contribute, how do I proceed?

Cheers,
Paul

ustccw, 2024-01-15:
@pauldeng Because AT HTTP related codes are hosted on a internal server, there is no a good way for you to contribute. thanks any way.

pauldeng, 2024-01-18:
Understood, thanks.

closed 2024-01-18 by pauldeng