espressif/esp-at #776: AT-SPI SDSPI with STM32 doesn't work

MichaelXlivnenko, 2023-11-03:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

HI,

I'm trying to port SDSPI example to STM32L4, but it doesn't work.

As you noted in README, I was starting with checking that hardware works well.
First that I do - upload ESP-AT to ESP32 as slave(using the latest version of core, esp-idf v.5.0.2 and ESP-WROOM-32 devkit). Then I upload ESP example from SDSPI folder and all works fine.("AT" command return "OK") 

Then I tried with STM32F103ZE dev board to check the STM32 example before porting and it doesn't work.

The connection is next:

STM32    --->   ESP32
SCLK (PA5)   --->    CLK(GPIO14)
MOSI (PA7)   --->    CMD(GPIO15)
MISO (PA6)   --->    DAT0(GPIO2)
GPIO_INT (PA0)   --->    DAT1(GPIO4)
CS (PA4)   --->    DAT3(GPIO13)

All lines excluding CLK is pulled-up with 10 kOhms resistors. 
Project was build using STM32CubeIDE from .ioc config file from example and all code is copied from example.

Debug console oputput:
> 10:24:14.414 -> E sdspi_trI sdspi_transaction: IOE: 0x02
10:24:16.191 -> I sdspi_transaction: IE: 0x03
10:24:16.191 -> I sdspi_transaction: BUS_WIDTH: 0x40
10:24:16.191 -> I sdspi_transaction: BUS_WIDTH: 0x60
10:24:16.191 -> I sdspi_transaction: Function 1 BS: 0000
10:24:16.191 -> I sdspi_transaction: Function 1 BS: 0200
10:24:16.191 -> D sdspi_transaction: spi_io_read_bytes, will transfer size: 4

and then all is stucked 

This is the datagramma of all lines after reset:
[image]

ustccw, 2023-11-03:
@ESP-Coco could you please help take a look

Jacques-Zhao, 2023-11-06:
@MichaelXlivnenko  Do you have more detailed LOG? I need to observe a step that detects a problem.

MichaelXlivnenko, 2023-11-06:
Did you mean LOG iutput in console or signals output?

Jacques-Zhao, 2023-11-06:
Full STM32 LOG, you can change deugLevel to 4(https://github.com/espressif/esp-at/blob/master/examples/at_spi_master/sdspi/STM32/Inc/port.h#L48)

MichaelXlivnenko, 2023-11-06:
10:27:55.666 -> V sdspi_transaction: CMD0
10:27:55.666 -> V sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0xffffffff