espressif/esp-at #779: AT+BLECONN? returns untrue data

fahirad, 2023-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My device is configured as a BLE client. 
The scenario is as follows:
I have one or more connected devices, let's say only one. Then I try to connect to another one, and during that process, during the `AT+BLECONN=<conn_index>,<remote_address>[,<addr_type>,<timeout>] ` command (its duration) if my already connected device gets disconnected, ESP will not pick it up, there will be no +BLEDISCONN message, nor will the AT+BLECONN? return true data (it will still return now disconnected device as connected).

P.S. My device is ESP32-C3_MINI-1, in case that is relevant and I am using the latest ESP-AT version:
AT+GMR output:

```
AT version:3.2.0.0(ec2dec2 - ESP32C3 - Jul 28 2023 07:04:22)
                  
SDK version:v5.0.2-376-g24b9d38a24
                                            
compile time(ed0e6d2c):Nov 13 2023 14:28:11
                                   
Bin version:v3.2.0.0(MINI-1)
```

ustccw, 2024-01-15:
@Luxin812 please check

Luxin812, 2024-03-20:
@fahirad thank you very much for asking this issue. I've reproduced it locally and will have a fix soon.

Luxin812, 2024-05-28:
This issue should be fixed in v3.3.0.0. Please verify in [v3.3.0.0](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Binary_Lists/esp_at_binaries.html) version. Thanks.

fahirad, 2024-07-02:
Hi @Luxin812 ,
I have tested this fix and it works now. Thanks a lot!

closed 2024-07-02 by fahirad