espressif/esp-at #788: Wi-Fi重连功能不生效

cha331, 2023-11-27:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

使用WebServer指令配网，发现配网完成是默认禁用自动重连的，调用`AT+CWRECONNCFG=10,0`，AP离线后重新上线，ESP32不会执行重连操作，断电重启后重连功能正常。
`[16:53:18.150]发→◇AT+CWRECONNCFG=10,0
□
[16:53:18.154]收←◆
OK

[16:53:41.944]发→◇AT+CWSTATE?
□
[16:53:41.949]收←◆+CWSTATE:2,"娓呯媯鐨刬Phone"

OK

[16:53:54.002]发→◇AT+CWJAP?
□
[16:53:54.008]收←◆+CWJAP:"娓呯媯鐨刬Phone","6a:ae:d5:ea:19:ed",6,-27,1,10,3,0,1

OK

[16:54:49.434]收←◆WIFI DISCONNECT
`

UART0输出如下，可以看到没有进行重连尝试：
`[16:40:20.624]收←◆dns server start
start web server

[16:52:23.295]收←◆connect config finish

[16:52:26.750]收←◆BLUFI BLE is not connected yet

[16:52:54.010]收←◆Delete Web Server
dns server stop

[16:54:49.435]收←◆wifi disconnected, rc:2

[16:55:19.245]收←◆No AP`

ustccw, 2023-12-05:
@cha331 感谢反馈。这是一个已知问题，内部已经修复过。请用附件里的库替换原有仓库 esp-at/components/at/lib 里的库验证。

[libespat.zip](https://github.com/espressif/esp-at/files/13560028/libespat.zip)

下一个版本 (v3.3.0.0) 将会包含这个修复。

cha331, 2023-12-06:
你好，替换掉库文件后重新编译，测试还是不行，webserver配网后wifi依然不会自动重连。

ustccw, 2024-01-15:
有替换成功吗，我在 master 上编译测试是可以的：
```
[16:04:10:153] ready
[16:04:12:773] AT+CWRECONNCFG=10,0
[16:04:12:773] 
[16:04:12:773] OK
[16:04:18:557] AT+CWMODE=3
[16:04:18:557] 
[16:04:18:557] OK
[16:04:20:506] AT+CWSAP="pos_softap","",11,0,3
[16:04:20:522] 
[16:04:20:522] OK
[16:04:27:855] AT+CIPMUX=1
[16:04:27:855] 
[16:04:27:855] OK
[16:04:28:766] AT+WEBSERVER=1,80,25
[16:04:28:782] 
[16:04:28:782] OK
[16:04:51:277] +STA_CONNECTED:"22:2c:5e:b6:01:70"
[16:04:51:516] +DIST_STA_IP:"22:2c:5e:b6:01:70","192.168.4.2"
[16:04:52:507] +DIST_STA_IP:"22:2c:5e:b6:01:70","192.168.4.2"
[16:05:02:716] +WEBSERVERRSP:1
[16:05:03:163] WIFI CONNECTED
[16:05:04:170] WIFI GOT IP
[16:05:05:097] +WEBSERVERRSP:2
[16:05:22:447] 
[16:05:22:447] ready
[16:05:22:511] WIFI CONNECTED
[16:05:23:501] WIFI GOT IP
[16:06:57:745] +STA_CONNECTED:"32:91:e2:d2:e4:76"
[16:06:59:183] +DIST_STA_IP:"32:91:e2:d2:e4:76","192.168.4.2"
[16:07:25:303] AT+GMR
[16:07:25:303] AT version:4.1.0.0-dev(571289c - ESP32C2 - Jan 12 2024 10:22:46)
[16:07:25:329] SDK version:v5.0.4-dirty
[16:07:25:329] compile time(a485dae9):Jan 15 2024 15:55:15
[16:07:25:329] Bin version:v4.0.0.0-dev(ESP32C2-4MB)
[16:07:25:329] 
[16:07:25:329] OK
```

你要不提供一下复现的 AT 命令，我试试看。

cha331, 2024-01-16:
> [16:38:11.675]收←◆
ready
[16:38:16.823]发→◇AT+CWRECONNCFG=10,0
□
[16:38:16.830]收←◆AT+CWRECONNCFG=10,0
OK
[16:38:21.877]发→◇AT+CWMODE=3
□
[16:38:21.883]收←◆AT+CWMODE=3
OK
[16:38:25.788]发→◇AT+CWSAP="pos_softap","",11,0,3
□
[16:38:25.794]收←◆AT+CWSAP="pos_softap","",11,0,3
OK
[16:38:29.582]发→◇AT+CIPMUX=1
□
[16:38:29.587]收←◆AT+CIPMUX=1
OK
[16:38:33.355]发→◇AT+WEBSERVER=1,80,25
□
[16:38:33.360]收←◆AT+WEBSERVER=1,80,25
OK
[16:39:30.471]收←◆+STA_CONNECTED:"d4:1b:81:4e:e5:07"
[16:39:30.605]收←◆+DIST_STA_IP:"d4:1b:81:4e:e5:07","192.168.4.2"
[16:40:04.759]收←◆+WEBSERVERRSP:1
[16:40:05.880]收←◆WIFI CONNECTED
[16:40:06.880]收←◆WIFI GOT IP
[16:40:08.275]收←◆+WEBSERVERRSP:2
[16:41:07.110]发→◇AT+WEBSERVER=0
□
[16:41:07.115]收←◆AT+WEBSERVER=0
OK
[16:41:19.064]发→◇AT+GMR
□
[16:41:19.069]收←◆AT+GMR
AT version:3.3.0.0-dev(e3a5abe - ESP32 - Dec  4 2023 10:15:36)
SDK version:v5.0.2-376-g24b9d38a24-dirty
compile time(282a4485):Jan 15 2024 16:19:03
Bin version:v3.2.0.0(MINI-1)
OK
[16:41:38.064]发→◇AT+CWMODE=1
□
[16:41:38.069]收←◆AT+CWMODE=1
+STA_DISCONNECTED:"d4:1b:81:4e:e5:07"
OK
[16:41:56.363]收←◆WIFI DISCONNECT

最后一条就是通过webserver配网的AP离线的提示信息，如果AP又上线了，也不会执行自动重连。必须要重启一次，自动重连才会生效。

lib文件确定是替换过且重新编译的，平台是ESP32-MINI-1
[image]

ustccw, 2024-01-16:
@cha331 昨天测试也复现了一种无法自动重连的情况，修复在这里：https://github.com/espressif/esp-at/commit/d65ee16f9ebc3d2390e865cf24fc7aa827f2e77d

可否重新拉取一下代码，试试看。

cha331, 2024-01-16:
感谢，拉取了master，测试确认问题已经修复。

closed 2024-01-16 by cha331