espressif/esp-at #789: Any Encrypted AT command feature available ?

vinodstanur, 2023-11-28:
### Is your feature request related to a problem?

From security point of view, plain text AT commands are not secure and not anyone can snif into it easily with almost zero effort.

### Describe the solution you'd like.

AT commands can be AES-encrypted, and a one-time key is randomly generated and exchanged for secure communication, adding an additional layer of encryption to make decoding more challenging for unauthorized entities.
Is is already available ? If not, any plan to implement it?

ustccw, 2023-12-05:
@vinodstanur Thank you for your feedback. We already have plans internally to implement this feature, as it's crucial for product security. The addition of this feature to AT is expected in 2024.Q1.

jcoteBrio, 2024-06-27:
Hi @ustccw, 

Can you confirm it is done since 2024.Q1 is now completed ?
If not when do you think this will be implemented ?

Best regards

ustccw, 2024-06-28:
I am really sorry for this delay. This feature would be implemented by an external example in 2024.Q3. 
Apologies for any inconvenience caused.

closed 2024-12-30 by igrr

ustccw, 2024-12-30:
@vinodstanur @jcoteBrio I'm very sorry for the long delay regarding this feature. The feature is now supported through the example. Could you please use [examples/at_interface_security](https://github.com/espressif/esp-at/tree/master/examples/at_interface_security) to verify it. If there is any questions, feel free to ask me.