espressif/esp-at #790: AT+SYSMFG: Improve with error code and HASH

MaJerle, 2023-12-05:
### Is your feature request related to a problem?

When using `AT+SYSMFG`, you get an error if you ask for more data to read than available, or if offset is larger than length of data. The same error is also if you simply put wrong parameters.

### Describe the solution you'd like.

It would be good to add 2 things in this command (and potentially `AT+SYSFLASH`, too):

- Add error codes that are returned based on the SYSMSG command. For example, length is too long, offset is too large, or something else.
- Add option to get `HASH/CRC32/...` of the MSG/FLASH content. In this case, we can check if flash content is the one we want, before we load new one. With this, we save the flash durability. A CRC32 is fast to compute on an MCU (even this HASH would be better tho, from security perspective) and easier to handle than to read full data back to MCU and compare if values are identiacal. 

### Describe alternatives you've considered.

There are no other alternatives, unfortunately. If command returns ERROR, you have to think what the problem is, usually end us w/o solution.

ustccw, 2023-12-06:
@MaJerle Thanks for the feedback. 
1) About adding error codes to `AT+SYSMFG`, early on, we were planning to add different error codes for each command. However, with the increase in firmware size (due to the addition of new AT features and the ESP-IDF upgrade), the firmware size for the ESP32 platform is nearly reaching its maximum user partition size, so we've become more cautious gradually.
Nevertheless, we have another significant plan to separate the logic of each AT command, deconstructing the code to reduce the bin size. Once this is accomplished, we will reconsider adding different error codes for each command.

2) Regarding the second feature, we already have a plan for it, and it might be completed by 2024.Q1.

~

mAliGem, 2026-08-19:
Just wanted to follow up on whether getting the HASH of MFG/FLASH content is still in the works?

I have created my own AT command in the meanwhile however, this is a very handy feature!

ustccw, 2026-08-20:
Hi @mAliGem ,
Sorry for the delay on this! Due to previous priorities, this feature was pushed back. We agree that having the MFG/FLASH hash check and corresponding error codes is very useful. We're planning to officially add this feature in Q3 2026.
Thanks for your patience and for sharing your workaround with us!