espressif/esp-at #791: "+BTDATA" is not received on classic bluetooth when the data is exactly or more than 20 symbols

ToniVankov, 2023-12-13:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.3.0.0-dev(s-da9d43f - ESP32 - Nov 30 2023 08:35:30)
SDK version:v5.0.4-dirty
compile time(b883f158):Dec 13 2023 10:07:21

### ESP-AT Firmware Source

Download from https://github.com/espressif/esp-at.git

### Hardware Information

ESP32-PICO-Mini-02

### Power Supply used

External 3.3V

### What is the expected behavior?

I expect it to receive "+BTDATA:" message for more than 20 symbols.

### What is the actual behavior?

If I send 20 or more symbols via bluetooth the ESP not provide the unsolicited "+DATA:" message.

### Probability of recurrence

For me this occur every time.

### AT+SYSRAM?

AT+SYSRAM?  
+SYSRAM:144532,138936
OK

### Steps to reproduce

AT+BTINIT=1
AT+BTSPPINIT=2
AT+BTNAME="ESP_PICO"
AT+BTSECPARAM=1,1,"1234"
AT+BTSCANMODE=2
AT+BTSPPSTART

Connecting via PC bluetooth:
(MAC is hidden here for security reasons)
+BTSPPCONN:0,"xx:xx:xx:xx:xx:xx"

Send one symbol via PC bluetooth:
+BTDATA:1,A

Send 30 symbols via PC bluetooth:
(nothing...)

### AT command port output

```plain
15:23: +BTSPPCONN:0,"xx:xx:xx:xx:xx:xx"
15:30: +BTDATA:3,A
```

### AT log port output

```plain
(no logs)
```

ustccw, 2024-01-15:
@Luxin812 please check.

Luxin812, 2024-03-20:
@ToniVankov Recently, the master branch has fixed similar problems. You can update your firmware to the latest and test again to see if your problem is also fixed.(https://github.com/espressif/esp-at/actions/runs/8294336044)

closed 2024-07-29 by ustccw