espressif/esp-at #794: Can not found confgen.py

ShiinaKaze, 2024-01-06:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html) and the issue is not addressed there.
- [X] I have compiled the hello_world example [Get Started with ESP-IDF](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#step-1-get-started-with-esp-idf) and the issue is not addressed there.
- [X] I have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### ESP-AT version

v2.2.1.0_esp8266-101-gf0436d77

### Operating System used

Windows

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

[image]

### What is the actual behavior?

[image]

### Steps to reproduce

./build.py menuconfig

### Build or installation Logs

```plain
Setting IDF_PATH environment variable: /home/ShiinaKaze/esp/esp-at/esp-idf
Checking Python dependencies...
Python requirements from /home/ShiinaKaze/esp/esp-at/esp-idf/requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory /home/ShiinaKaze/esp/esp-at/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp8266 -DCCACHE_ENA
BLE=0 /home/ShiinaKaze/esp/esp-at"...
silence:1 sdkconfig:C:/msys64/home/ShiinaKaze/esp/esp-at/module_config/module_esp8266_default/sdkcon
fig_silence.defaults
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.42.0")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/msys64/opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc.exe - skipp
ed
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++.exe - ski
pped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v2.2.1.0_esp8266-101-gf0436d77
-- Building ESP-IDF components for target esp8266
/usr/bin/python.exe: can't open file '/home/ShiinaKaze/esp/esp-at/build/C:/msys64/home/ShiinaKaze/es
p/esp-at/esp-idf/tools/kconfig_new/confgen.py': [Errno 2] No such file or directory
CMake Error at esp-idf/tools/cmake/kconfig.cmake:178 (message):
  Failed to run confgen.py
  (C:/msys64/usr/bin/python.exe;C:/msys64/home/ShiinaKaze/esp/esp-at/esp-idf/tools/kconfig_new/confg
en.py;--kconfig;C:/msys64/home/ShiinaKaze/esp/esp-at/esp-idf/Kconfig;--config;C:/msys64/home/ShiinaK
aze/esp/esp-at/sdkconfig;--defaults;C:/msys64/home/ShiinaKaze/esp/esp-at/module_config/module_esp826
6_default/sdkconfig_silence.defaults;--env-file;C:/msys64/home/ShiinaKaze/esp/esp-at/build/config.en
v).
  Error 2
Call Stack (most recent call first):
  esp-idf/tools/cmake/build.cmake:423 (__kconfig_generate_config)
  esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:55 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
idf.py build ret: 2
idf.py build failed
```

ustccw, 2024-01-15:
@ShiinaKaze Have you finished the steps in the [Get Started Guide](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/v3.4/get-started/index.html) before you start your ESP-AT.

ShiinaKaze, 2024-01-15:
> @ShiinaKaze Have you finished the steps in the [Get Started Guide](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/v3.4/get-started/index.html) before you start your ESP-AT.

Yes, I have done it.

ustccw, 2024-01-16:
There should be a [confgen.py](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/tools/kconfig_new/confgen.py) file in your esp-idf/tools/kconfig_new/confgen.py. Could you please check it.

you can try to delete the whole esp-idf directory and let the esp-at clones the esp-idf again.
```
rm -rf esp-idf
python build.py build
```

ShiinaKaze, 2024-01-16:
> There should be a [confgen.py](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/tools/kconfig_new/confgen.py) file in your esp-idf/tools/kconfig_new/confgen.py. Could you please check it. [...]

Thanks for your reply, I have checked it.
[image]
But the problem is still there.

ustccw, 2024-01-16:
Just run the commands and test again.
```
rm -rf esp-idf
python build.py build
```

Is it possible that it is caused by encryption software, firewalls, and read/write permissions, or something else.

closed 2025-03-17 by ustccw