espressif/esp-at #798: 怎么打开模块回应https的错误码

ctzxc88, 2024-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

在文档中我只看到了错误码的定义但没看到怎么开启这个错误码？我尝试在app-main中使用esp_log_level_set("＊",ESP_LOG_);但也没有作用，我想问两个问题：
1:怎么通过编程的方法而不是menucongfig的方法将所有模块或者至少https client请求的错误码和错误日志打开？（目前使用menuconfig有bug但程序能编译成功）
2:目前使用https请求模块回应ASCII文:  ERROR  0000000056488,  是什么意思？

ctzxc88, 2024-01-26:
就像我在上面说的那样目前menuconfig有问题无法使用，目前使用直接编程的方法解决问题。设备平台是esp32c3往我的mcu传输数据，我是使用官方的at指令httpclient指令执行get请求下载数据，它会同时通过spi传输给mcu,但下载数据到0x19bbbb以及更多时报错：
block _is_free(prev) && "prev block is not free through marked as "
是不是spi驱动内存回收时出的错？因为我通过esplog打印了httpclient的perform函数执行貌似没有报错。

ustccw, 2024-02-26:
日志的话，修改对应模组配置(module_config 目录下) 里的 sdkconfig.defaults 文件：
```
# 移除行
# CONFIG_LOG_DEFAULT_LEVEL_ERROR=y

# 增加行
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
```

HTTP 和 SPI 代码是开源的，增加一些日志调试看看。如果调试有困难，请提供复现代码，详细的复现步骤，以及完整的日志。

closed 2024-07-29 by ustccw