espressif/esp-at #799: ESP-AT will not support ESP32-S

vanvught, 2024-01-11:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html) and the issue is not addressed there.
- [X] I have compiled the hello_world example [Get Started with ESP-IDF](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#step-1-get-started-with-esp-idf) and the issue is not addressed there.
- [X] I have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### ESP-AT version

v4.0.0.0-6-ga42a3b65

### Operating System used

macOS

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

According to [https://espressif-docs.readthedocs-hosted.com/projects/esp-at/en/release-v2.3.0.0_esp32c3/Compile_and_Develop/How_to_implement_SPI_AT.html](https://espressif-docs.readthedocs-hosted.com/projects/esp-at/en/release-v2.3.0.0_esp32c3/Compile_and_Develop/How_to_implement_SPI_AT.html) 
> This document mainly introduces the implementation and use of SPI AT for ESP32-C and ESP32-S series.

So I would expect that ./build.py menuconfig is showing the platform ESP32-S series

### What is the actual behavior?

```
[@MacBook-Air esp-at (master)]$ ./build.py menuconfig
Platform name:
1. PLATFORM_ESP32C6
2. PLATFORM_ESP32C2
3. PLATFORM_ESP32
4. PLATFORM_ESP32C3
choose(range[1,4]):
```
There is no **PLATFORM_ESP32S2**

### Steps to reproduce

```
mkdir esp
cd esp
git clone --recursive https://github.com/espressif/esp-at.git
cd esp-at
./build.py menuconfig
```

### Build or installation Logs

```plain
N.A.
```

vanvught, 2024-01-11:
According to the comments in https://github.com/espressif/esp-at/commit/921d7db7ac7e854d6d3c23e839c44048db05538c only the support for ESP8266 should be removed.
@ustccw

ustccw, 2024-05-17:
`ESP32-S2` is supported on the master branch. The final firmware is expected to be released in July 2024.

closed 2024-10-10 by ustccw