espressif/esp-at #800: Connect ESP32-C3 to STM32 (IDFGH-11893)

shlomo-edu, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I installed the ESP-AT binary on my ESP32-C3-DevKit1 and tested it from a YAT terminal.
It works great.
Then I connected its module to my STM32 via TX and RX pins:
   PD2     ------> USART5_RX : GPIO_PIN_2
   PC10    ------> USART4_TX : GPIO_PIN_5
Now I send a command: AT+GMR from the STM via the UART using the following line:
`HAL_UART_Transmit_IT(&huart4, buff, Len);`
And I expect to get the result (same result that I see when I connected the terminal via the DevKit) in the callback function:
`HAL_UART_RxCpltCallback`
But I don't get anything.
Also, I don't understand why I can't place a breakpoint in the callback function.
I am using IAR workbench IDE, and I can place breakpints everywhere, but not in the callbacks.

ustccw, 2024-02-26:
Regarding the issue of not being able to set breakpoints, perhaps you could inquire on the STM32 forum or STM32 related channel, where you might find a quicker solution.

shlomo-edu, 2024-02-26:
I solved all my problems with integrationg  ESP32-C3 in an STM32 PCB.
It works OK now.
Thanks.

closed 2024-02-26 by shlomo-edu