espressif/esp-at #801: AT+BLEGATTSNTFY无反应

monkey-qiu, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我现在使用模式：
从设备：ESP32-C3，使用 AT 固件
主设备：ESP32-C3，基于 ble50_gattc_gattx_coex 编译的固件
现在住设备连接到从设备后，会每隔100ms向从设备发送数据，从设备在收到数据后会进行答复，现在在答复的时候调用AT+BLEGATTSNTFY指令没反应，或者报ERROR，从串口1抓错误信息为write data warning, len:0

jamesxwx, 2024-01-16:
@monkey-qiu  提供一下你使用的 AT 把呢不能信息

AT+GMR

jamesxwx, 2024-01-16:
@monkey-qiu  你改一下命令的顺序，每次发送 AT+BLEGATTSNTFY 之后顺便查询一下剩余内存 AT+SYSRAM?

jamesxwx, 2024-01-17:
@monkey-qiu   ESP-AT  只支持 三个 连接，不管是主机还是从机，累计总和 3 个连接

monkey-qiu, 2024-01-18:
> @monkey-qiu 你改一下命令的顺序，每次发送 AT+BLEGATTSNTFY 之后顺便查询一下剩余内存 AT+SYSRAM?

这个指令也不响应，现象是出现busy by....后就所有AT指令都不响应了

monkey-qiu, 2024-01-18:
> @monkey-qiu 你改一下命令的顺序，每次发送 AT+BLEGATTSNTFY 之后顺便查询一下剩余内存 AT+SYSRAM?

 环境：
从设备：ESP32-C3，使用 AT 固件，一直处于AT模式，外接mcu，作为server
主设备：ESP32-C3，基于 ble50_gattc_gattx_coex 编译的固件作为client
测试流程：client连接上server后，每隔50ms向server发送数据，server在接收到数据后通过串口把数据透传给mcu，mcu通过AT+BLEGATTSNTFY把原数据返回给server，server把数据发送给clinet，数据长度80字节

Luxin812, 2024-03-20:
AT 产生 busy py 的原因：
[image]
可以先检查下你的测试逻辑上条指令是否执行完了，可以使用串口工具手动测试看下能否复现问题。

closed 2024-07-29 by ustccw