espressif/esp-at #804: at+cwjap命令无法连接到wifi，不是返回连接超时，就是返回密码错误

jack-k737, 2024-01-27:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g24b9d38a24-dirty
compile time(6118fc22):Jul 28 2023 09:47:28

### ESP-AT Firmware Source

https://dl.espressif.com/esp-at/firmwares/esp32/ESP32-WROOM-32/ESP32-WROOM-32-AT-V3.2.0.0.zip

### Hardware Information

ESP32-WROOM-32E

### Power Supply used

USB

### What is the expected behavior?

正常连接到路由器或者手机热点

### What is the actual behavior?

使用AT+CWJAP="ssid","password"指令，不是返回连接超时就是返回密码错误，即使ssid和password都没有错误

### Probability of recurrence

must appear

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:166096,147832
OK

### Steps to reproduce

1.通过usb转ttl，将ESP32-WROOM-32E的串口（GPIO16和GPIO17）还有gnd，连到电脑
2,使用串口助手，输入at+cwjap指令

### AT command port output

```plain
AT+CWJAP="xiejun","xiej12345678"
+CWJAP:1

ERROR
AT+CWJAP="xiejun\0","xiej12345678\0"
+CWJAP:3

ERROR
AT+CWLAP
+CWLAP:(3,"xiejun",-48,"5e:b1:86:92:aa:44",6,-1,-1,4,4,7,0)
+CWLAP:(4,"TP-LINK_506",-53,"cc:34:29:67:fc:64",1,-1,-1,4,4,7,1)
+CWLAP:(3,"406",-58,"30:5f:77:7e:66:31",11,-1,-1,4,4,7,1)
+CWLAP:(4,"CMCC-b2yz",-68,"00:0e:1f:7e:a0:37",10,-1,-1,5,3,7,1)
+CWLAP:(4,"TP-LINK_711",-73,"94:d9:b3:35:2f:c3",1,-1,-1,4,4,7,0)
+CWLAP:(3,"",-79,"54:f2:94:5c:f7:fd",11,-1,-1,4,4,7,0)
+CWLAP:(4,"CMCC-1001F",-79,"54:f2:94:5c:f7:f8",11,-1,-1,4,4,7,1)
+CWLAP:(4,"runningteddy",-80,"c0:a5:dd:98:70:70",6,-1,-1,4,4,7,1)
+CWLAP:(4,"CMCC-DAvX",-80,"9c:7f:81:df:f3:11",11,-1,-1,4,4,7,1)
+CWLAP:(4,"",-84,"fa:8c:21:aa:81:f5",6,-1,-1,4,4,7,0)
+CWLAP:(4,"D-Link_DIR-629",-85,"9c:d6:43:ce:18:c8",6,-1,-1,5,3,7,1)
+CWLAP:(3,"xsgd311",-86,"8c:de:f9:55:41:6f",11,-1,-1,4,4,7,1)
+CWLAP:(0,"",-91,"a4:fa:76:ac:48:6b",1,-1,-1,0,0,7,0)

OK
AT+CWLAP?

ERROR
AT+CWLAP
+CWLAP:(3,"406",-46,"30:5f:77:7e:66:31",11,-1,-1,4,4,7,1)
+CWLAP:(3,"xiejun",-52,"5e:b1:86:92:aa:44",1,-1,-1,4,4,7,0)
+CWLAP:(4,"TP-LINK_506",-54,"cc:34:29:67:fc:64",1,-1,-1,4,4,7,1)
+CWLAP:(4,"CMCC-b2yz",-75,"00:0e:1f:7e:a0:37",10,-1,-1,5,3,7,1)
+CWLAP:(3,"",-77,"54:f2:94:5c:f7:fd",11,-1,-1,4,4,7,0)
+CWLAP:(4,"",-78,"9e:7f:81:4f:f3:11",11,-1,-1,4,4,7,0)
+CWLAP:(4,"runningteddy",-79,"c0:a5:dd:98:70:70",6,-1,-1,4,4,7,1)
+CWLAP:(4,"CMCC-DAvX",-79,"9c:7f:81:df:f3:11",11,-1,-1,4,4,7,1)
+CWLAP:(4,"TP-LINK_711",-80,"94:d9:b3:35:2f:c3",1,-1,-1,4,4,7,0)
+CWLAP:(3,"",-80,"54:f2:94:5c:f7:f9",11,-1,-1,4,4,7,0)
+CWLAP:(4,"LYX",-83,"68:77:24:17:14:de",6,-1,-1,4,4,7,1)
+CWLAP:(4,"",-83,"7e:77:24:17:14:de",6,-1,-1,4,4,7,0)
+CWLAP:(3,"DIRECT-HKLAPTOP-PSRGOIABmsLK",-89,"ba:9a:2a:5f:be:f0",11,-1,-1,4,4,6,1)
+CWLAP:(4,"Xiaomi_EEE8",-90,"a4:39:b3:ad:ee:e9",11,-1,-1,5,3,7,1)
+CWLAP:(4,"TP-LINK_81F5",-91,"f8:8c:21:ba:81:f5",6,-1,-1,4,4,7,1)
+CWLAP:(3,"",-91,"ba:80:35:49:89:77",6,-1,-1,4,4,7,0)

OK
AT+CWJAP="406","xf000712"
+CWJAP:1

ERROR
AT+CWJAP?

OK

ready
AT+CWMODE?
+CWMODE:1

OK
AT+CWJAP="xiejun","xiej12345678"
+CWJAP:1

ERROR
AT+GMR
AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g24b9d38a24-dirty
compile time(6118fc22):Jul 28 2023 09:47:28
Bin version:v3.2.0.0(WROOM-32)

OK
AT+CWJAP="xiejun","xiej12345678"
+CWJAP:1

ERROR
```

### AT log port output

```plain
????
```

### More Information.

除了最新版本，上一个版本也试过了，还有一开始出厂的版本，均无法连接wifi，但是ap模式可以用

jack-k737, 2024-01-30:
刚烧录的固件mac地址为0x0的无效地址，需要手动修改mac地址

ustccw, 2024-02-26:
发布的 v3.2.0.0 AT 固件，经过多轮测试、成千上万个 testcases，覆盖了多种路由器，AP。因此 AT 固件本身不太可能出问题。

连接不上有多种可能，比如：
1. 硬件供电问题（用万用表测一下 vcc 和 gnd，确认一下是不是 3.3v）
2. AP 兼容性问题。从日志看，`xiejun` 这个 AP 出现了 1 信道和 6 信道。需要抓取一下 802.11 包，看下问题出现在哪里。
3. 硬件本身问题（硬件受损过，比如高压击穿了某些元器件，强磁过，等等）。这种换一个硬件确认一下即可。

jack-k737, 2024-02-26:
> > “刚烧录的固件 mac 地址为 0x0 的无效地址，需要手动修改 mac 地址”
> 
> 没有很理解这句的意思，能否详细说明一下您的具体操作。

可能我的板子硬件出了问题吧，就是通过at指令查询板子的mac地址，发现是0.0.0.0.0.0这样的无效地址。继续通过at指令手动修改mac地址，就能联网了。

jack-k737, 2024-02-26:
> > “刚烧录的固件 mac 地址为 0x0 的无效地址，需要手动修改 mac 地址”
> 
> 没有很理解这句的意思，能否详细说明一下您的具体操作。

可能固件启动的时候，一开始读取硬件mac地址出现了问题，或者硬件的mac本身就是0吧。小白的猜测

ustccw, 2024-02-26:
比较奇怪。可以使用 [esptool.py](https://github.com/espressif/esptool) 工具来查看 efuse 里的 mac 地址，这个 mac 地址是从 IEEE 获取到的全球唯一的 mac 地址。

[image]

closed 2024-03-18 by ustccw