espressif/esp-at #806: ESP32-WROVER-32-AT-V2.4.0.0 firmware cannot connect to Azure IOT hub.

zh0903, 2024-02-15:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Espressif Team,

I flash newest ESP-AT firmware into my ESP32-WROVER board 
I run following AT command as https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/AT_Command_Set/MQTT_AT_Commands.html :

AT+CIPSNTPCFG=1,8,"pool.ntp.org"
AT+CIPSNTPTIME?
AT+MQTTUSERCFG=0,3,"attest","hansHub.azure-devices.net/attest/?api-version=2021-04-12","",0,0,""
AT+MQTTPASSWORD=0,"MY_SAS_TOKEN"
AT+MQTTCONN=0,"hansHub.azure-devices.net",8883,1

Get error in the last step. Same configuration works in MQTTX/MQTTBox tool. Here is the output of serial tool:
[image]

Would you like to told me:
1. If ESP-AT firmware can connect to(or support) Azure IOT hub?
2.Is there any way to output detail client/device side logs of AT-command?

Thanks!

zh0903, 2024-02-27:
Thanks, let me try it on.

ustccw, 2024-03-18:
@zh0903 Hello!
Regarding the first point: You are correct. AT has not formally released the `v3.2.0.0` firmware for ESP32-WROVER-32, but you can download firmware close to `v3.2.0.0` according to [docs](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_download_the_latest_temporary_version_of_AT_from_github.html).

Regarding the second point:
We suggest using `v3.2.0.0` or later versions. You can update configurations such as certificates and private keys in one of the following ways:
- Use the [AT+SYSMFG](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-sysmfg) command to dynamically update certificates and private keys.
- Use the [at.py Tool](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/tools_at_py.html) script to statically update certificates and private keys.
- Update certificates and private keys directly during compilation, refer to [How to Update PKI Configuration](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_update_pki_config.html) for more details.

zh0903, 2024-03-18:
Thanks for your instruction!