espressif/esp-at #810: AT+MDNS does not produce the expected record

SeraphimlabsLLC, 2024-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

AT versions 3.1 and higher do not appear to produce the expected mdns record. The host portion is being replaced by the service name. It is likely that this applies as far back as 2.2, I've only tested 3.1 and 4.1. 

With a working AT 1.7.4, the command string AT+MDNS=1,"dcchost","withrottle",2560
results in avahi-browse --all showing 
+ enp41s0 IPv4 dcchost                                       _withrottle._tcp     local

If I use AT 3.1 or 4.1 with the same AT+MDNS=1,"dcchost","withrottle",2560, 
avahi-browse --all actually breaks with avahi_service_browser_new() failed: Invalid service type, this is actually a known bug with this tool when a malformed broadcast was made recently. 

Using mdns-scan instead, I see the malformed 
+ withrottle.withrottle._tcp.local  instead. 

Please advise what can be done. I've tried a couple different ESP-AT versions on an ESP32-C3-WROOM-2 module, compiling from the git repository in order to assign the pin numbers correctly.

ustccw, 2024-05-21:
@SeraphimlabsLLC many thanks for your report, i agreed. 

**on AT v1.7.4,  esp-at uses hostname as the instance name:**
[image]

**on AT v3.3.0, esp-at uses the service name as the instance name:**
[image]

maybe we should add one `AT+CIPMDNS` command to configure the parameters amap for mdns.
I'll let you know once this feature is done. thanks again!

CW

ustccw, 2024-10-10:
@SeraphimlabsLLC sorry for the late reply. we have already fixed this. AT+MDNS added more parameters for it. could you please have a try. 
- [AT+MDNS command](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/TCP-IP_AT_Commands.html#at-mdns-configure-the-mdns-function)
- [mDNS example](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/TCP-IP_AT_Examples.html#example-mdns)