espressif/esp-at #814: Cannot build due to WARNING: Component "espressif/mdns" not found

veso266, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello there, I have NodeMCU v1.1 board (it says: ESP-WROOM-32 on the metal part)
[image]

and I was trying to build and load this firmware on it

I have this development enviroment (the latest) on my Windows maschine: https://dl.espressif.com/dl/idf-installer/esp-idf-tools-setup-offline-5.2.1.exe?
and I can compile blink example just fine

I use command line I do not use espressif IDE

I used this instructions to try to compile ESP-AT: https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#esp-at-idf-prerequisites

I used the same path as in instructions
after running python build.py install I selected option 1 (ESP-32 on WROOM-32) and again Option 1 to not print any log files

everything went fine until I ran 
```
python build.py menuconfig
```

Then I get this error: WARNING: Component "espressif/mdns" not found
[image]

and it quits

why does this happen, do I use to new enviroment?
What do I do?

Thanks for Anwsering and Best Regards

ustccw, 2024-03-18:
@veso266 It seems that the relevant components were not successfully downloaded.
Please delete the `build sdkconfig` and run the following commands again, providing the complete log afterwards.
```
python build.py install
python build.py menuconfig
```

veso266, 2024-03-18:
thanks, at the end I just installed the version 5 of the toolchain because I read in the readme that this is the version that the software was ported to: https://dl.espressif.com/dl/idf-installer/esp-idf-tools-setup-offline-5.0.exe?

and it worked fine, so I guess the latest version is to new for this to build

ustccw, 2024-03-18:
Good to know! thanks for the feedback.

closed 2024-03-18 by ustccw