espressif/esp-at #815: AT+BLESCAN error, with name/mac filter. release/4.0.0.0 with ble command support

F1RST1, 2024-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]

repo: https://github.com/F1RST1/esp-at

F1RST1, 2024-03-20:
chip type: esp32-c2

F1RST1, 2024-03-25:
Hello. @xcguang @ustccw 
Could you guys help me?

Luxin812, 2024-03-27:
@F1RST1 thank you very much for asking this issue. I've reproduced it locally and will have a fix soon.

F1RST1, 2024-04-08:
@Luxin812 
 By the way, as I know, the implementation of filtering by name is fully comparison. Could yout please change it to partical compare? I meant using function like `strstr` instead of `strcmp`.

Luxin812, 2024-04-09:
The latest master branch has fixed the problem of BLE returning ERROR when scanning the specified name and mac. You can download the latest firmware to test it.

```By the way, as I know, the implementation of filtering by name is fully comparison. Could yout please change it to partical compare? I meant using function like strstr instead of strcmp.```
Maybe you can use the AT+BLESCAN command without filtering. Use the AT+SYSMSGFILTERCFG command to filter. You can study this command:
Command reference: https://docs.espressif.com/projects/esp-at/en/latest/esp32c2/AT_Command_Set/Basic_AT_Commands.html#at-sysmsgfiltercfg-query-set-the-system-message-filters
Example reference: https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/at_msg_resp_fmt_ctrl_examples.html#at-response-message-format-control-examples

closed 2024-07-26 by F1RST1