espressif/esp-at #816: Is it possible to disable some GATTS using at.py Tool GATTS Configuration?

hamonika00, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I would like to use the at.py Tool to disable GATTS Configuration after some numbers. What should I do?
I use ESP32-WROOM-32 Series. thank you

ustccw, 2024-05-17:
@hamonika00 Hello, you can refer to [Modify GATTS Configuration](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/tools_at_py.html#at-py-modify-gatts) for more details.

hamonika00, 2024-05-20:
> @hamonika00 Hello, you can refer to [Modify GATTS Configuration](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/tools_at_py.html#at-py-modify-gatts) for more details.

hello, I can 'Modify' GATTS Configuration over ay.py tool. I want GATTS to be deleted so it doesn't appear.

Luxin812, 2024-07-01:
Currently, the GATTS configuration cannot be deleted through the at.py tool, only it can be modified. If you want to delete part of the GATTS configuration, you can refer to the documentation: https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_customize_BLE_services.html

Luxin812, 2024-07-02:
If you want to delete modifications or some services, you can study the [AT+SYSMFG](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#at-sysmsg-query-set-system-prompt-information) command and [customized_partitions/raw_data/ble_data/gatts_data.csv](https://github.com/espressif/esp-at/blob/8b01b678/components/customized_partitions/raw_data/ble_data/gatts_data.csv).
```
[16:48:00:456] at+bleinit=2␍␊
[16:48:01:560] ␍␊
[16:48:01:560] OK␍␊
[16:48:02:828] AT+BLEGATTSSRVCRE␍␊
[16:48:03:333] ␍␊
[16:48:03:333] OK␍␊
[16:48:04:393] AT+BLEGATTSSRVSTART␍␊
[16:48:04:403] ␍␊
[16:48:04:403] OK␍␊
[16:48:05:724] AT+BLEGATTSSRV?␍␊
[16:48:05:739] +BLEGATTSSRV:1,1,0xA002,1␍␊
[16:48:05:739] +BLEGATTSSRV:2,1,0xA003,1␍␊
[16:48:05:739] ␍␊
[16:48:05:739] OK␍␊
[16:48:08:034] AT+BLEGATTSCHAR?␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,1,0xC300,0x02␍␊
[16:48:08:074] +BLEGATTSCHAR:"desc",1,1,1,0x2901␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,2,0xC301,0x02␍␊
[16:48:08:074] +BLEGATTSCHAR:"desc",1,2,1,0x2901␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,3,0xC302,0x08␍␊
[16:48:08:074] +BLEGATTSCHAR:"desc",1,3,1,0x2901␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,4,0xC303,0x04␍␊
[16:48:08:074] +BLEGATTSCHAR:"desc",1,4,1,0x2901␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,5,0xC304,0x08␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,6,0xC305,0x10␍␊
[16:48:08:074] +BLEGATTSCHAR:"desc",1,6,1,0x2902␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,7,0xC306,0x20␍␊
[16:48:08:074] +BLEGATTSCHAR:"desc",1,7,1,0x2902␍␊
[16:48:08:074] +BLEGATTSCHAR:"char",1,8,0xC307,0x02␍␊
[16:48:08:105] +BLEGATTSCHAR:"desc",1,8,1,0x2901␍␊
[16:48:08:105] +BLEGATTSCHAR:"char",2,1,0xC400,0x02␍␊
[16:48:08:105] +BLEGATTSCHAR:"desc",2,1,1,0x2901␍␊
[16:48:08:105] +BLEGATTSCHAR:"char",2,2,0xC401,0x02␍␊
[16:48:08:105] +BLEGATTSCHAR:"desc",2,2,1,0x2901␍␊
[16:48:08:105] ␍␊
[16:48:08:105] OK␍␊
[16:49:15:150] AT+SYSMFG=1,"ble_data","cfg18"␍␊
[16:49:15:175] +SYSMFG:"ble_data","cfg18",7,24,18,16,0x2803,0x01,1,1,20␍␊
[16:49:15:175] ␍␊
[16:49:15:175] OK␍␊
[16:49:27:125] **AT+SYSMFG=0,"ble_data","cfg18"**␍␊
[16:49:27:140] ␍␊
[16:49:27:140] OK␍␊
[16:49:32:739] AT+BLEGATTSCHAR?␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,1,0xC300,0x02␍␊
[16:49:32:781] +BLEGATTSCHAR:"desc",1,1,1,0x2901␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,2,0xC301,0x02␍␊
[16:49:32:781] +BLEGATTSCHAR:"desc",1,2,1,0x2901␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,3,0xC302,0x08␍␊
[16:49:32:781] +BLEGATTSCHAR:"desc",1,3,1,0x2901␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,4,0xC303,0x04␍␊
[16:49:32:781] +BLEGATTSCHAR:"desc",1,4,1,0x2901␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,5,0xC304,0x08␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,6,0xC305,0x10␍␊
[16:49:32:781] +BLEGATTSCHAR:"desc",1,6,1,0x2902␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,7,0xC306,0x20␍␊
[16:49:32:781] +BLEGATTSCHAR:"desc",1,7,1,0x2902␍␊
[16:49:32:781] +BLEGATTSCHAR:"char",1,8,0xC307,0x02␍␊
[16:49:32:813] +BLEGATTSCHAR:"desc",1,8,1,0x2901␍␊
[16:49:32:813] +BLEGATTSCHAR:"char",2,1,0xC400,0x02␍␊
[16:49:32:813] +BLEGATTSCHAR:"desc",2,1,1,0x2901␍␊
[16:49:32:813] +BLEGATTSCHAR:"char",2,2,0xC401,0x02␍␊
[16:49:32:813] +BLEGATTSCHAR:"desc",2,2,1,0x2901␍␊
[16:49:32:813] ␍␊
[16:49:32:813] OK␍␊
[16:49:39:861] AT+RESTORE␍␊
[16:49:41:631] ␍␊
[16:49:41:631] OK␍␊
[16:49:42:722] ␍␊
[16:49:42:722] ready␍␊
[16:49:47:556] at+bleinit=2␍␊
[16:49:48:633] ␍␊
[16:49:48:633] OK␍␊
[16:49:50:624] AT+BLEGATTSSRVCRE␍␊
[16:49:50:918] ␍␊
[16:49:50:918] OK␍␊
[16:49:52:251] AT+BLEGATTSSRVSTART␍␊
[16:49:52:259] ␍␊
[16:49:52:259] OK␍␊
[16:50:00:070] AT+BLEGATTSSRV?␍␊
[16:50:00:071] +BLEGATTSSRV:1,1,0xA002,1␍␊
[16:50:00:087] ␍␊
[16:50:00:087] OK␍␊
[16:50:02:071] AT+BLEGATTSCHAR?␍␊
[16:50:02:115] +BLEGATTSCHAR:"char",1,1,0xC300,0x02␍␊
[16:50:02:115] +BLEGATTSCHAR:"desc",1,1,1,0x2901␍␊
[16:50:02:115] +BLEGATTSCHAR:"char",1,2,0xC301,0x02␍␊
[16:50:02:115] +BLEGATTSCHAR:"desc",1,2,1,0x2901␍␊
[16:50:02:115] +BLEGATTSCHAR:"char",1,3,0xC302,0x08␍␊
[16:50:02:115] +BLEGATTSCHAR:"desc",1,3,1,0x2901␍␊
[16:50:02:115] +BLEGATTSCHAR:"char",1,4,0xC303,0x04␍␊
[16:50:02:115] +BLEGATTSCHAR:"desc",1,4,1,0x2901␍␊
[16:50:02:115] +BLEGATTSCHAR:"char",1,5,0xC304,0x08␍␊
[16:50:02:115] +BLEGATTSCHAR:"char",1,6,0xC305,0x10␍␊
[16:50:02:115] +BLEGATTSCHAR:"desc",1,6,1,0x2902␍␊
[16:50:02:115] ␍␊
[16:50:02:115] OK␍␊
```

closed 2024-10-10 by ustccw