espressif/esp-at #819: Writing GATT characteristic with variable size will result in characteristc being truncated

jnstahl, 2024-03-29:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:4.1.0.0-dev(6eebeb4 - ESP32C6 - Mar 20 2024 12:32:44)
SDK version:v5.1.2-dirty
compile time(77e1c56):Mar 27 2024 07:39:19
Bin version:v4.1.0.0-dev(ESP32C6-4MB)

### ESP-AT Firmware Source

Download from Github Actions

### Hardware Information

ESP32-C6-MINI1

### Power Supply used

External 3.3V

### What is the expected behavior?

I am writing characteristics from an Android app. The characteristic size may be of variable length for successive writes. With ESP-AT 2.4.0 and ESP32C3 it was working all correctly. With ESP32C6 and firmware 4.0.0 I was expecting it to behave the same way

### What is the actual behavior?

With the ESP432C6 and Firmware 4.0.0 (also 4.1.0.0-dev) the behavior is changed. Apparently the firmware will limit some internal buffer to match the initial chararacteristic length. So when writing the characteristc with 14 bytes, then subsequent writes will have the bytes after the 14th byte corrupted, even though the +WRITE message still indiciates the longer size.

### Probability of recurrence

appears under all conditions

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:259140,250072
OK

### Steps to reproduce

1. Initialize BLE server and start advertising
2. connect nrfConnect 
3. from nrfConnect, write Characteristic 0xC04 a first time with one length (e.g. "ABCDEFGHIJKL")
4. from nrfConnect, write Characteristic 0xC04 a second time with a larger length (e.g. "ABCDEFGHIJKLMNOPQRSTUVWXYZ")
5. RESULT -> the second time the characteristic is limited to the length established in the first write

### AT command port output

```plain
AT+GMR
AT version:4.1.0.0-dev(6eebeb4 - ESP32C6 - Mar 20 2024 12:32:44)
SDK version:v5.1.2-dirty
compile time(77e1c56):Mar 27 2024 07:39:19
Bin version:v4.1.0.0-dev(ESP32C6-4MB)
OK
AT+BLEINIT=2
OK
AT+BLEADVPARAM=50,50,0,0,7,0,,
OK
AT+BLEADVDATA="0201060A0954696d6544726f7073030302A0"
OK
AT+BLEADVSTART
OK
+BLECONN:0,"4d:ad:0d:10:66:5e"
[... 245 lines cut ...]
registering descriptor 0x2901 with handle=37
registering characteristic 0xc401 with def_handle=38 val_handle=39
registering descriptor 0x2901 with handle=40
reset reason=0
connection established; status=0 handle=1 our_ota_addr_type=0 our_ota_addr=4084b47f:00:15:40828000:00:4217d198
 our_id_addr_type=0 our_id_addr=4084b471:00:15:40828000:00:4217d198
 peer_ota_addr_type=1 peer_ota_addr=4084b486:00:15:40828000:00:4217d198
 peer_id_addr_type=1 peer_id_addr=4084b478:00:15:40828000:00:4217d198
 conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0
advertise complete; reason=0connection updated; status=0 handle=1 our_ota_addr_type=0 our_ota_addr=4084b42f:00:15:40828000:00:4217d198
 our_id_addr_type=0 our_id_addr=4084b421:00:15:40828000:00:4217d198
 peer_ota_addr_type=1 peer_ota_addr=4084b436:00:15:40828000:00:4217d198
 peer_id_addr_type=1 peer_id_addr=4084b428:00:15:40828000:00:4217d198
 conn_itvl=6 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

connection updated; status=0 handle=1 our_ota_addr_type=0 our_ota_addr=4084b42f:00:15:40828000:00:4217d198
 our_id_addr_type=0 our_id_addr=4084b421:00:15:40828000:00:4217d198
 peer_ota_addr_type=1 peer_ota_addr=4084b436:00:15:40828000:00:4217d198
 peer_id_addr_type=1 peer_id_addr=4084b428:00:15:40828000:00:4217d198
 conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0
```

### More Information.

The workaround is to pad the characteristc for each write to a fixed (maximum length) value. However this is using significantly more bandwidht

ustccw, 2024-05-17:
@Luxin812 PTAL

Luxin812, 2024-05-28:
Hi @jnstahl:

This issue should have been fixed on the latest master branch. Please download the AT firmware from the latest master branch and perform verification testing.
```
[12:10:29:235] AT+BLEINIT=2
[12:10:29:538] 
[12:10:29:538] OK
[12:10:30:799] AT+BLEADVPARAM=50,50,0,0,7,0,,
[12:10:30:801] 
[12:10:30:801] OK
[12:10:32:265] AT+BLEADVDATA="0201060A0954696d6544726f7073030302A0"
[12:10:32:286] 
[12:10:32:286] OK
[12:10:33:688] AT+BLEADVSTART
[12:10:33:692] 
[12:10:33:692] OK
[12:10:42:862] +BLECONN:0,"64:2a:9a:5b:7e:64"
[12:10:44:411] +BLECONNPARAM:0,6,6,6,0,500
[12:10:45:466] +BLECONNPARAM:0,40,40,40,0,500
[12:10:58:534] +WRITE:0,1,5,,18,cgjhfgnvxfbbccfvgg
[12:11:14:047] +WRITE:0,1,5,,31,fhhdvnhccvbbggggggggggvgvvggggg
[12:11:39:241] +WRITE:0,1,5,,32,gnvcbbfcvnbccbbbggvvcfvbvffvvvvg
[12:12:01:990] +WRITE:0,1,5,,14,asdfghjklasdfg
[12:13:07:699] +WRITE:0,1,5,,26,asdfghjklasdfghjklasdfghjk
[12:13:13:358] AT+GMR
[12:13:13:380] AT version:3.4.0.0-dev(be2bd9b - ESP32C6 - May 17 2024 05:26:09)
[12:13:13:380] SDK version:v5.1.2-dirty
[12:13:13:380] compile time(a24d4fb4):May 27 2024 11:29:31
[12:13:13:380] Bin version:v4.1.0.0-dev(ESP32C6-4MB)
[12:13:13:380] 
[12:13:13:380] OK
```

jnstahl, 2024-06-01:
@Luxin812 checked, works now.

closed 2024-06-01 by jnstahl