espressif/esp-at #825: Is the WT32-S1 still supported?

shawty, 2024-04-17:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a wireless-tag combo networking board, I noticed that the firmware on it is dated 2019 (Version 1.2 I think)

I'm trying to work out, if I can compile ands use the latest versions of the AT firmware on this device.

The only information I can get from the device when it boots is follows:

```
I (29) boot: ESP-IDF 5d27367-dirty 2nd stage bootloader
I (29) boot: compile time 20:25:30
I (29) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 0000d000 00004000
I (65) boot:  1 phy_init         RF data          01 01 00011000 00001000
I (72) boot:  2 factory          factory app      00 00 00020000 00200000
I (80) boot:  3 abup_fota        OTA data         01 00 00220000 00100000
I (87) boot: End of partition table
I (91) boot: Defaulting to factory image
I (96) ABUP: abup_bl_init_flash: app_offset = 0x20000, app_size = 0x200000

I (104) ABUP: abup_bl_init_flash: ota_offset = 0x220000, ota_size = 0x100000

I (111) ABUP: Delta file is not exist, exit.
I (116) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x3c6cc (247500) map
I (212) esp_image: segment 1: paddr=0x0005c6f4 vaddr=0x3ffbdb60 size=0x0391c ( 14620) load
I (218) esp_image: segment 2: paddr=0x00060018 vaddr=0x400d0018 size=0xeb954 (964948) map
I (557) esp_image: segment 3: paddr=0x0014b974 vaddr=0x3ffc147c size=0x00340 (   832) load
I (557) esp_image: segment 4: paddr=0x0014bcbc vaddr=0x40080000 size=0x00400 (  1024) load
I (564) esp_image: segment 5: paddr=0x0014c0c4 vaddr=0x40080400 size=0x175e8 ( 95720) load
I (626) boot: Loaded app from partition at offset 0x20000
I (626) boot: Disabling RNG early entropy source...
I (627) cpu_start: Pro cpu up.
I (630) cpu_start: Application information:
I (635) cpu_start: Project name:     ethernet_basic
I (640) cpu_start: App version:      5d27367-dirty
I (646) cpu_start: Compile time:     Nov 26 2019 20:24:44
I (652) cpu_start: ELF file SHA256:  b3b686af6ffc043e...
I (658) cpu_start: ESP-IDF:          5d27367-dirty
I (663) cpu_start: Starting app cpu, entry point is 0x4008113c
I (0) cpu_start: App cpu up.
I (674) heap_init: Initializing. RAM available for dynamic allocation:
```

I managed at one point, to get the device into programming mode, but all I got was the following

```
light_timer start ok
ABUP-->Please wait 1 sec.
ABUP-->Please wait 2 sec.
ABUP-->Please wait 3 sec.
ABUP-->Please wait 4 sec.
ABUP-->Please wait 5 sec.
ABUP-->Fota partition: type=1, subtype=0x00, address=0x220000, size=0x100000, label="abup_fota", encrypted=0
ABUP-->No Upgrade! version=1.2
ABUP-->abup_get_pid_psec fail, device should register.
ABUP-->Current progress 0 ################
I (6407) user_eth: eth_event_handler
I (6407) user_eth: Ethernet Link Up
I (7807) tcpip_adapter: eth ip: 192.168.19.113, mask: 255.255.128.0, gw: 192.168.17.2
I (7807) user_eth: Ethernet Got IP Address
I (7807) user_eth: ~~~~~~~~~~~
E (7807) event: system event loop not initialized via esp_event_loop_init
I (7807) user_eth: ETHIP:192.168.19.113
I (7817) user_eth: ETHMASK:255.255.128.0
I (7827) user_eth: ETHGW:192.168.17.2
I (7827) user_eth: ~~~~~~~~~~~
ABUP-->DNS lookup succeeded. IP=0.0.0.0
ABUP-->... allocated socket
ABUP-->... socket connect failed errno=113
ABUP-->abup_close_socket
ABUP-->... allocated socket
ABUP-->... socket connect failed errno=113
ABUP-->abup_close_socket
ABUP-->... allocated socket
ABUP-->... socket connect failed errno=113
ABUP-->abup_close_socket
ABUP-->... allocated socket
ABUP-->... socket connect failed errno=113
ABUP-->abup_close_socket
ABUP-->... allocated socket
```

Can you tell me if it's possible to put the current version of the AT firmware on this WT32-S1 device, and if so, which file do I need to download and where from.

Regards
Shawty

ustccw, 2024-05-17:
@shawty Hi, may be you can seek help from your module supplier, or you may consider [purchasing modules from Espressif](https://www.espressif.com/en/products/modules).

shawty, 2024-05-17:
Hi I did, no reply.

But I think it's a simple question for you, does the AT firmware in this repository work for the device I mention.  The device is based on an espressif chip set, and it does have what appears to be an older version on.

JAndrassy, 2024-05-17:
which ESP32 chip does esptool report?

shawty, 2024-05-17:
that's a good question, the device doesn't have a USB on it the same as a regular ESP32, you can hook 2 TTL Uarts up to the device once appears to be a system console (And supposedly an Arduino IDE compatible programming port) and the second appears to be the "AT interface" used by the device.

This is the company that makes the device : https://en.wireless-tag.com/

and this is the device that I'm trying to use : https://en.wireless-tag.com/product-item-2.html

I tried running ESPTool against both Uart's and it didn't want to know, but the ESP Flash tool GUI thing, downloaded direct from espessif does connect to the first Uart, it just doesn't tell me anything about the device.

There's a partial of the "Boot log" in my first post in this topic, that and a bit more get's emitted from the first UART when the device is powered up, I don't recall seeing any kind of CPU-ID in that boot log, but I can run it again over the weekend (It's currently back in it's box at the moment)  and give you the full boot output here.

All the documentation says is that it's an ESP32 compatible device.

JAndrassy, 2024-05-17:
it has to be in bootloader mode for esptool

shawty, 2024-05-17:
As far as I was aware when I tired it, it was, as I say though, I'll dig it out of it's box over the weekend if I get chance and confirm.